Koenkk/zigbee2mqtt #12139: Lidl Thermostat TS0601 not showing any logs

Flipsoo, 2022-04-11:
### What happened?

After a while working without any problems 3 of my 4 thermostats stopped working. There is no success with repairing because z2m doesnt get any informations.
Tried with herdsman_debug and all debug logs i could find but i cant see anything. 

Steps taken:
- reinstall zigbee2mqtt 
- Restored old backup
- Removed thermostat tried to readd - no logs
- Tried to re-pair - no logs

### What did you expect to happen?

Thermostat should show interview or reconnected

### Zigbee2MQTT version

1.25.0-dev commit: 422b7e0

### Adapter firmware version

20210708

### Adapter

Sonoff usb 3 zigbee dongle

### Debug log

Im sorry to not provide any logs but i cant find anything than the saved entity entrie .

smulle48, 2022-04-11:
Have you added any new device lately to your network?

I had problem with my Danalock V3 when tried to configure the pincode. It "killed" my zigbee2mqtt network and nothing really worked until I removed the lock from the network and restarted z2m. Then I re-add'ed the lock without configure the pincode and everything worked again

Flipsoo, 2022-04-11:
hey thanks for your answer, will try to delete a few devices, but there was no real new device joining the network.
i also tried to restore the whole system to a earlier working state out of january and still there is no signal detected by z2m.

Flipsoo, 2022-04-11:
hey thanks for that tipp to, i tried to update my sonoff dongle to the version [...]launchpad_coordinator_20220219, but that also doesnt help. the point that bothers me the most is that one of my 4 same brand same charge thermostat is still working, is it possible that they refuse to connect to my network? they are in the config and database but the state doesnt update at all.... force_remove doesnt help either, there is still no log or signal comming in

Flipsoo, 2022-04-12:
As a workaround i setup a fresh instance of zigbee2mqtt edge, changed the `panID` to a different value, reflashed the Sonoff Zigbee USB Stick again with the `[...]launchpad_coordinator_20220219` firmware and restarted zigbee2mqtt and after that i was able to rejoin all thermostats without any issues sure this required me to repair all devices in my flat but i had the change to properly rename them all.

I dont know what had happend but resetting everything to start helped.