Koenkk/zigbee2mqtt #12149: Moes ZP-LZ-FR2U L1 and L2 toggling together

fsevilla3, 2022-04-12:
### What happened?

The [Moes ZP-LZ-FR2U](https://www.zigbee2mqtt.io/devices/ZP-LZ-FR2U.html) socket plugs expose two switch endpoints: `state_l1` for the AC outlet and `state_l2` for 2 USB outlets. However, toggling either switch automatically changes the state of the other. This problem happens from Home Assistant and directly from the Zigbee2MQTT UI.

### What did you expect to happen?

I expected the switch endpoints to be controllable independently of the other.

### How to reproduce it (minimal and precise)

Toggle either switch endpoint and observe the state of the other changing as well.

### Zigbee2MQTT version

1.25.0-dev commit: [2aa1b4e](https://github.com/Koenkk/zigbee2mqtt/commit/2aa1b4e)

### Adapter firmware version

20220219

### Adapter

SONOFF controller running zStack3x0

### Debug log

```
zigbee2mqtt      | Zigbee2MQTT:info  2022-04-12 15:21:00: MQTT publish: topic 'zigbee2mqtt/Moes Socket Plug Girls Room', payload '{"linkquality":61,"power_outage_memory":"restore","state_l1":"OFF","state_l2":"ON"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-04-12 15:21:00: MQTT publish: topic 'zigbee2mqtt/Moes Socket Plug Girls Room', payload '{"linkquality":61,"power_outage_memory":"restore","state_l1":"OFF","state_l2":"OFF"}'
```

Koenkk, 2022-04-12:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_cymsnfvf'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_2xlvlnez'}],
    model: 'ZP-LZ-FR2U',
    vendor: 'Moes',
    description: 'Zigbee 3.0 dual USB wireless socket plug CUSTOM',
    fromZigbee: [fz.on_off, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
        exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage')],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(device.getEndpoint(1));
        await reporting.onOff(device.getEndpoint(2));
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

fsevilla3, 2022-04-12:
Hi @Koenkk,

Thanks for the quick reply. I have tested your change and it works perfectly. I'm happy to test again in dev once you've merged your change. Really appreciate your work on Zigbee2MQTT.

Kind regards,
Federico

Koenkk, 2022-04-13:
Pushed the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-13 by Koenkk

fsevilla3, 2022-04-14:
Awesome stuff @Koenkk. Tested latest dev branch and it works perfectly.

duvalale, 2023-09-17:
Sorry for reopening this issue, but I have the exact same device, and it is recognized as [Nue / HGZB-01](https://www.zigbee2mqtt.io/devices/HGZB-01.html).
Does it still work for you ?

I'm with the following config : 
Version de Zigbee2MQTT : 1.33.0
Coordinator : zStack3x0 (zzh!)
Firmware version : 20221226

Thanks