Koenkk/zigbee2mqtt #12151: [Feature request]: TI SimpleLink SDK 5.40.00.40 based Zigbee Router Z-Stack firmware?

Hedda, 2022-04-12:
### Is your feature request related to a problem? Please describe

Can the Zigbee Router firmware please be updated to TI SimpleLink SDK 5.40.00.40 or later?

https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0

https://software-dl.ti.com/simplelink/esd/simplelink_cc13xx_cc26xx_sdk/5.40.00.40/exports/changelog.html

### Describe the solution you'd like

Binary release of Z-Stack 3.x.0_20220125 router firmware images based on  TI SimpleLink SDK 5.40.00.40 or later:

https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0

### Describe alternatives you've considered

The alternative is to wait or use maybe build a PTVO Zigbee Configurable Firmware but not use which TI SDK version it uses.

### Additional context

Koenkk's latest released Zigbee Router firmware is based on TI SimpleLink SDK 5.30.01.01 

https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/CHANGELOG.md

Koenkk, 2022-04-12:
The next TI release should be soon, waiting for that currently.

Hedda, 2022-05-15:
Still wanted

Hedda, 2022-06-17:
> The next TI release should be soon, waiting for that currently.

Looks like TI have released it now as SimpleLink cc13xx_cc26xx "v6x" series now as version 6.10.00.29 (as of 8th of April 2022) or?

https://www.ti.com/tool/download/SIMPLELINK-CC13XX-CC26XX-SDK

https://www.ti.com/tool/download/SIMPLELINK-CC13XX-CC26XX-SDK/6.10.00.29

https://software-dl.ti.com/simplelink/esd/simplelink_cc13xx_cc26xx_sdk/6.10.00.29/exports/changelog.html

https://software-dl.ti.com/simplelink/esd/simplelink_cc13xx_cc26xx_sdk/6.10.00.29/exports/release_notes_simplelink_cc13xx_cc26xx_sdk_6_10_00_29.html

closed 2022-07-26 by github-actions[bot]