Koenkk/zigbee2mqtt #12152: Legrand FC80CC: unable to set back atuto mode"

pirracas77, 2022-04-12:
### What happened?

When the device is in switch mode, it cannot be set up back to auto mode via MQTT. It only works pressing the fisical button.

You can go from auto-mode to switch mode via MQTT but you cannot go from switch-mode to auto-mode via mqtt.

### What did you expect to happen?

go from switch-mode to auto-mode via mqtt.

### How to reproduce it (minimal and precise)

Set the swith-mode via web page. Try to return to auto-switch mode using the same way. Nothing happends in the device

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20210120

### Adapter

slaesh cc2652rb

### Debug log

Debug Received MQTT message on 'zigbee2mqtt/0x000474000083fbee/set' with data '{"state":"ON"}'
Debug Publishing 'set' 'state' to '0x000474000083fbee'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"auto","last_seen":"2022-04-12T11:51:49+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"auto","last_seen":"2022-04-12T11:51:49+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"ON"}'
Debug Received MQTT message on 'zigbee2mqtt/0x000474000083fbee/set' with data '{"state":"OFF"}'
Debug Publishing 'set' 'state' to '0x000474000083fbee'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"auto","last_seen":"2022-04-12T11:51:51+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"auto","last_seen":"2022-04-12T11:51:51+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'
Debug Received MQTT message on 'zigbee2mqtt/0x000474000083fbee/set' with data '{"device_mode":"switch"}'
Debug Publishing 'set' 'device_mode' to '0x000474000083fbee'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"auto","last_seen":"2022-04-12T11:51:54+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"switch","last_seen":"2022-04-12T11:51:54+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'
Debug Received Zigbee message from '0x000474000083fbee', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"switch","last_seen":"2022-04-12T11:51:54+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'
Debug Received MQTT message on 'zigbee2mqtt/0x000474000083fbee/set' with data '{"state":"ON"}'
Debug Publishing 'set' 'state' to '0x000474000083fbee'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"switch","last_seen":"2022-04-12T11:51:56+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"switch","last_seen":"2022-04-12T11:51:56+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"ON"}'
Debug Received MQTT message on 'zigbee2mqtt/0x000474000083fbee/set' with data '{"state":"OFF"}'
Debug Publishing 'set' 'state' to '0x000474000083fbee'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"switch","last_seen":"2022-04-12T11:51:56+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"switch","last_seen":"2022-04-12T11:51:56+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'
Debug Received MQTT message on 'zigbee2mqtt/0x000474000083fbee/set' with data '{"device_mode":"auto"}'
Debug Publishing 'set' 'device_mode' to '0x000474000083fbee'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"switch","last_seen":"2022-04-12T11:51:58+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0x000474000083fbee', payload '{"device_mode":"auto","last_seen":"2022-04-12T11:51:58+02:00","led_when_on":"OFF","linkquality":141,"permanent_led":"OFF","power":0,"state":"OFF"}'

closed 2022-05-26 by github-actions[bot]

celestianx, 2022-11-14:
If you are interested, I have on my way to bring support for this feature to z2m @pirracas77

pirracas77, 2022-12-02:
> If you are interested, I have on my way to bring support for this feature to z2m @pirracas77

Great!! I see your PR included in last version. I will test it.