Koenkk/zigbee2mqtt #12153: [New device support]: Zigbee Smart Plans Watering Timer

MoonshineSG, 2022-04-12:
### Link

https://www.aliexpress.com/item/1005004073061424.html

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0xXXXX","nwkAddr":22628,"manufId":4417,"manufName":"_TZE200_arge1ptm","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0018X�)f\u0019X�)\u0012","65506":54,"65508":1,"stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff07df2e","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1649755156956,"defaultSendRequestWhen":"immediate"}

### Comments

Hi managed to make it "supported" by adding a basic external  converter, and got all the Data Points from Tuya website, but I kinda failed at what's  next... I read the "how to", but got lost...

Some of the DP are in Chinese and I did a google translate. Kinda makes sense ... 

```
1 - Switch
2 - Regulating water volume
3 - Flow state
10 - Weather Delay
11 - Irrigation time

101 - 倒计时剩余时间  countdown time remaining
102 - 倒计时剩余时间设置 countdown remaining time setting
103 - 开到底状态 open to the end
104 - 故障告警 fault alarm
105 - 默认倒计时开启 by default countdown is on
106 - 默认倒计时设置 default countdown settings
107 - 月使用时长 monthly usage time
108 - 月使用水容量 monthly water capacity
109 - 定时灌溉 regular irrigation
110 - 电池电量 battery power
```

Also this info is available on the Tuya site 

```
switch | Boolean | "{true,false}"
percent_control | Integer | {   "unit": "%",   "min": 0,   "max": 100,   "scale": 0,   "step": 5 }
weather_delay | Enum | {   "range": [     "cancel",     "24h",     "48h",     "72h"   ] }
countdown | Integer | {   "unit": "s",   "min": 0,   "max": 86400,   "scale": 0,   "step": 1

```

Any help will be appreciated. 

### External converter

```shell
fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_arge1ptm'
        },
    ],
    model: 'QT-05M',
    vendor: 'Moes',
    description: 'Smart Water Timer',
```

MoonshineSG, 2022-04-14:
Managed to add some of the datapoints and I can receive the status of the device, but it can't send any "commands"... Hopefully I can get some help here... 

latest converter:

```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

// 1 - Switch

// 2 - Regulating water volume (set)
// 3 - Flow state (read)

// 10 - Weather Delay
// 11 - Irrigation time

// 101 - 倒计时剩余时间  countdown time remaining (read)
// 102 - 倒计时剩余时间设置 countdown remaining time setting (set)

// 103 - 开到底状态 open to the end
// 104 - 故障告警 fault alarm
// 105 - 默认倒计时开启 by default countdown is on
// 106 - 默认倒计时设置 default countdown settings

// 107 - 月使用时长 monthly usage time
// 108 - 月使用水容量 monthly water capacity

// 109 - 定时灌溉 regular irrigation
// 110 - 电池电量 battery power

const tuyaLocal = {
    dataPoints: {
        percent_control: 2 ,
        percent_state: 3,
        
        battery: 110,
    },
};

const fzLocal = {
    watering_timer: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport'],
        convert: (model, msg, publish, options, meta) => {

            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp; // First we get the data point ID
                const value = tuya.getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                switch (dp) {
                    case tuyaLocal.dataPoints.percent_control: {
                        result.percent_control = value
                        break;
                    }
                    case tuyaLocal.dataPoints.percent_state: {
                        result.percent_state = value
                        break;
                    }
                    case tuyaLocal.dataPoints.battery: {
                        // value reported in percent
                        result.battery = value;
                        break;
                    }
                    default: {
                        meta.logger.debug(`>>> UNKNOWN DP #${dp} with data "${JSON.stringify(dpValue)}"`);
                    }
                }
            }

            meta.logger.debug(`******* watering_timer >> from zigbee: "${JSON.stringify(result)}"`);

            return result;
        },
    },
};

const tzLocal = {
    percent_control: {
        key: ['percent_control'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointRaw(entity, tuyaLocal.dataPoints.percent_control, tuya.convertDecimalValueTo4ByteHexArray(value));
        },
    },   
};

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_arge1ptm'}],
    model: 'QT-05M',
    vendor: 'QOTO?',
    description: 'Solar watering timer',
    fromZigbee: [fz.ignore_basic_report, fz.ignore_tuya_set_time, fz.ignore_onoff_report, fzLocal.watering_timer],
    toZigbee: [
        tzLocal.percent_control,
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.battery(),
        exposes.numeric('percent_state', ea.STATE).withUnit('%').withDescription('Current water flow'),
        exposes.numeric('percent_control', ea.STATE_SET).withValueMin(0).withValueMax(100).withUnit('%').withDescription('Flow control'),
    ],
};

module.exports = definition;
```

MoonshineSG, 2022-04-14:
Changed `tzLocal `  to            
```
 const percentage = Math.round(value / 10) * 10; 
 await tuya.sendDataPointValue(entity, tuyaLocal.dataPoints.percent_control, percentage);

```

and I have control. 👍🏻👍🏻 Now adding the scheduling commands...

zamog, 2022-05-19:
@MoonshineSG Did you manage to get it work ?

balistof, 2022-05-20:
> I managed a partial support of the main functions and settings. I tried to contact the manufacturer but they are not very forth coming with info, so I'll just leave it as it is, since it is in very usable state.

Hi @MoonshineSG is the external converter somewhere available? Would be great if you can share it

zamog, 2022-05-20:
+1 I just got one, will be great to add zigbee2mqtt support base on it

MoonshineSG, 2022-05-20:
I am almost done with moving to a new place, so I'll find some time to publish the codes. Do note that it is my first draft, and for now works for the specific model.
 
I'll allow contributors, so maybe it will get polished in time...

MoonshineSG, 2022-05-29:
@balistof, @zamog, you can get the codes from https://github.com/MoonshineSG/zigbee2mqtt-custom/

Add the `QT-05M.js` to your `config/zigbee2mqtt/ `folder and load it in the UI...

Please not that this is a first working draft, and I had no time to polish it too much... Hope it does the work for now. 

Send me feedback... .

balistof, 2022-05-29:
first tests and it works perfect. tested manual, automatic with timer and homeassistant integration. all good!

zamog, 2022-05-31:
@MoonshineSG - Works, thanks! 
Two notes:
1. Can you reopen this issue so it will be add to the next release
2. I hit an error when running an automation with timer value > 100 - i think we need to add .withValueMax(14400) to exposes.numeric('timer', ea.STATE_SET) - WDYT ?

Koenkk, 2022-06-24:
@MoonshineSG if you are interested, please create a pr for zigbee-heredsman-converters to add support for this device out-of-the-box.

abuhabeeb, 2022-07-08:
Its work like Sharm with QT-05M.js.  Thank You.

I hope it can support for this device out-of-the-box.

closed 2022-08-15 by github-actions[bot]