Koenkk/zigbee2mqtt #12158: ROBB Smarrt 230V warm/koud wit LED Trafo Zigbee ROB_200-061-0

mrblwd-058, 2022-04-12:
### Link

https://www.robbshop.nl/robb-smarrt-230v-warm-koud-wit-led-trafo-zigbee?sqr=zigbee%20led%20driver

### Database entry

{"id":34,"type":"Router","ieeeAddr":"0x9035eafffe97be94","nwkAddr":3216,"manufId":4644,"manufName":"ROBB  smarrt","powerSource":"Mains (single phase)","modelId":"ROB_200-061-0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"ROB_200-061-0","manufacturerName":"ROBB  smarrt","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r60"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r60","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1649782004047}

### Comments

I didn't try something myself, I've seen the instructions but my knowledge isn't that good. Thanks a lot for supporting this device!

Koenkk, 2022-04-12:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

mrblwd-058, 2022-04-12:
After selecting the LcolorCtrl the "read" button is greyed out.

mrblwd-058, 2022-04-12:
Let me know if you need more information! thanks

Koenkk, 2022-04-13:
Could you check if it works with the following external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ROB_200-061-0'],
    model: 'ROB_200-061-0',
    vendor: 'ROBB',
    description: 'Zigbee LED dimmer',
    extend: extend.light_onoff_brightness_colortemp(),
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if it works

mrblwd-058, 2022-04-13:
Hi Koen, I can turn the device on/off and the brightness slider is available etc. but the lights are not going on. I don't know if that is because a configuration mismatch or something else. 
[image]

Any Idea?

Koenkk, 2022-04-13:
Could you confirm the device works with another Zigbee bridge (if you have one available)?

mrblwd-058, 2022-04-13:
I only have a USB attached Zigbee receiver on my HA server.

Koenkk, 2022-04-13:
Maybe you can try using ZHA and confirm if works there, I want to rule out any malfunctioning from the device side. (since the on/off commands are standard I would expect it to respond to that)

mrblwd-058, 2022-04-13:
I bought 3 of these so I configure another one with a simple bulb LED dimmable light. Keep u posted.

mrblwd-058, 2022-04-13:
I just see in Z2M "Availability
 [Disabled]" didn't mentioned that before, what does that mean?

mrblwd-058, 2022-04-13:
Robbshop sent me the wrong devices, I figured out that this is a LED Driver for CCT (LED Strips) I send an e-mail for a solution, hopefully they have the device I need for my spots. I guess that one need also to be supported hehe :)

closed 2022-05-21 by github-actions[bot]