Koenkk/zigbee2mqtt #12162: Exception in async_discover when dispatching mqtt_discovery_new_light_mqtt

half-monkey, 2022-04-12:
### What happened?

Added a Philips Hue bulb to the ZigBee network, the bulb does not show up in Home Assistant, and there is an error in the Home Assistant log which implies some of the JSON being sent is not quite what Home Assistant expects:

2022-04-12 22:18:49 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Bathroom_Shower/set', 'device': {'identifiers': ['zigbee2mqtt_0x001788010c9e1a29'], 'manufacturer': 'Philips', 'model': 'Hue White and Color Ambiance GU10 (929001953101)', 'name': 'Bathroom_Shower', 'sw_version': '1.65.11_hB798F2B'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'max_mireds': 500, 'min_mireds': 150, 'name': 'Bathroom_Shower', 'schema': 'json', 'state_topic': 'zigbee2mqtt/Bathroom_Shower', 'supported_color_modes': ['xy', 'color_temp'], 'unique_id': '0x001788010c9e1a29_light_zigbee2mqtt', 'platform': 'mqtt'},)
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 152, in async_discover
config = schema(discovery_payload)
File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__
return self._exec((Schema(val) for val in self.validators), v)
File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec
raise e if self.msg is None else AllInvalid(self.msg, path=path)
File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec
v = func(v)
File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
return schema(data)
File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 25, in validate_mqtt_light
return schemas[value[CONF_SCHEMA]](value)
File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
return base_validate(path, iteritems(data), out)
File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['color_mode']

and similar entries relating to the same bulb and other Hue bulbs.
Control of the bulbs appears to work well in zigbee2mqtt's web interface.

### What did you expect to happen?

The bulb is discovered normally and appears as a light in Home Assistant

### How to reproduce it (minimal and precise)

 - run zigbee2mqtt in a Docker container, connected to an MQTT server (Mosquitto in my case)
 - add logger:
  default: warning
  logs:
    homeassistant.components.mqtt: info
to configuration.yaml for Home Assistant
 - run Home Assistant 2021.3.4 in a Docker container connected to the same MQTT server
 - Add a Philips Hue bulb to the ZigBee network using the web interface of zigbee2mqtt
 - watch it not appear in home assistant, and observe home assistant logs.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20210120

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

half-monkey, 2022-04-12:
I haven't included the zigbee2mqtt log, as there doesn't appear to be any problem in zigbee2mqtt (bulbs all work perfectly through the web interface) - it looks like just a problem with the JSON being sent to Home Assistant to me, but if you really need the zigbee2mqtt log, let me know and I'll create one.

half-monkey, 2022-04-12:
Relevant config message in MQTT looks like: {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/Bathroom_Shower/set","device":{"identifiers":["zigbee2mqtt_0x001788010c9e1a29"],"manufacturer":"Philips","model":"Hue White and Color Ambiance GU10 (929001953101)","name":"Bathroom_Shower","sw_version":"1.65.11_hB798F2B"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"max_mireds":500,"min_mireds":150,"name":"Bathroom_Shower","schema":"json","state_topic":"zigbee2mqtt/Bathroom_Shower","supported_color_modes":["xy","color_temp"],"unique_id":"0x001788010c9e1a29_light_zigbee2mqtt"}

Koenkk, 2022-04-13:
Try updating to the latest HA version. You are using a one year old version.

closed 2022-05-21 by github-actions[bot]