Koenkk/zigbee2mqtt #12165: Sonoff USB Dongle plus with firmware CC1352P2_CC2652P_launchpad_coordinator_20220219 can't find any device

osan15, 2022-04-13:
### What happened?

Hello,

I have a problem with the Sonoff dongle (model ZBDongle-P).
I installed the firmware CC1352P2_CC2652P_launchpad_coordinator_20220219.
When I launch the Zigbee2MQTT web interface, I don't have any ZigBee device appearing in the list.
I have several devices from Xiaomi's Aqara range, 2 devices from Heiman (HS1MS-EF, HS2WD-E).

Could you give me more information about this problem or if I made a mistake during the installation.

Thanks in advance

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220219

### Adapter

ZBDongle-P

johnnykang, 2022-04-14:
have you paired them with your dongle?

osan15, 2022-04-20:
Sorry for my late reply. Indeed the problem was the pairing of the device. For Heirman devices, you have to keep pressing the reset button for about 10 seconds to switch to pairing mode.

Is it possible to update the documentation for HS1MS-EF, HS2WD-E.
I can do the PR if necessary.

Thanks for your help.

closed 2022-05-29 by github-actions[bot]