Koenkk/zigbee2mqtt #12167: keen smart vents 612 1.1 [New device support]:

ecto1a2003, 2022-04-13:
### Link

https://keenhome.io/

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0x0022a3000011fbd8","nwkAddr":29380,"manufId":4443,"manufName":"Keen Home Inc","powerSource":"Battery","modelId":"SV01-612-MP-1.1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":3,"inClusterList":[0,1,3,4,5,6,8,32,1026,1027,2821,64513,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SV01-612-MP-1.1","manufacturerName":"Keen Home Inc","powerSource":3,"zclVersion":2,"appVersion":19,"stackVersion":51,"dateCode":"20151116R100219"}},"genPollCtrl":{"attributes":{"checkinInterval":240}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00212effff081445","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":19,"stackVersion":51,"dateCode":"20151116R100219","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1649882846868,"defaultSendRequestWhen":"fastpoll"}

### Comments

intimidating, seems to be a firmware revision difference between this and others

ecto1a2003, 2022-04-13:
i made this, seems to work.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SV01-612-MP-1.1'],
    model: 'SV01',
    vendor: 'Keen Home',
    description: 'Smart vent',
    fromZigbee: [fz.cover_position_via_brightness, fz.temperature, fz.battery, fz.keen_home_smart_vent_pressure,fz.ignore_onoff_report],
    toZigbee: [tz.cover_via_brightness],
    meta: {battery: {dontDividePercentage: true}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['genLevelCtrl', 'genPowerCfg', 'msTemperatureMeasurement', 'msPressureMeasurement'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.temperature(endpoint);
        await reporting.pressure(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.cover_position().setAccess('state', ea.ALL), e.temperature(), e.battery(), e.pressure()],
};

module.exports = definition;

ecto1a2003, 2022-04-14:
i dont know what to do with this 

Debug Received Zigbee message from 'Foyer Vent', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":19,"imageType":0,"manufacturerCode":4443}' from endpoint 1 with groupID null
Debug Device 'Foyer Vent' requested OTA

Koenkk, 2022-04-14:
Added the device.

> Debug Received Zigbee message from 'Foyer Vent', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":19,"imageType":0,"manufacturerCode":4443}' from endpoint 1 with groupID null
Debug Device 'Foyer Vent' requested OTA

This can be ignored, it is the device that requests an OTA update, but we don't have OTAs available for this device.

closed 2022-04-14 by Koenkk