Koenkk/zigbee2mqtt #12169: Zigbee switches becomes non responsive after a short time

jyavenard, 2022-04-14:
### What happened?

I got two new zigbee switches :
a 3 gangs:
https://www.amazon.com.au/ZigBee-Smart-Switch-Automation-Control/dp/B0969YLP5T/ref=sr_1_6?crid=GCWXZJ6YFZB2&keywords=zigbee+switch+2-way&qid=1649916751&sprefix=zigbee+switch+2-way%2Caps%2C226&sr=8-6
and 1 gang:
https://www.amazon.com.au/ZigBee-Smart-Switch-Automation-Control/dp/B09T2MDG2Z/ref=sr_1_8?crid=GCWXZJ6YFZB2&keywords=zigbee+switch+2-way&qid=1649916775&sprefix=zigbee+switch+2-way%2Caps%2C226&sr=8-8

They are seen by zigbee2mqtt as [HGZB-43](https://www.zigbee2mqtt.io/devices/HGZB-43.html#nue%2520%252F%25203a-hgzb-43) and
[HGZB-41](https://www.zigbee2mqtt.io/devices/HGZB-41.html#nue%2520%252F%25203a-hgzb-41) 
respectively.

They look and feel the same as the earlier generation HGZB-41 and HGZB-43 except that one gang on a switch can be paired with the gang of another switch to enable 2-way.
That is you turn on a gang on one switch, and the paired gang on the other switch will automatically turn on as well. No need for a zigbee controller, that always work and respond instantly.

They are paired with zigbee2mqtt as usual.
Since upgrading to zigbee2mqtt 1.25.0 however, I can set the state of those switch only for about 5 minutes after power-cycling the switch or re-pairing them.
After that, you can't set the state, but you can read it.

So if I physically turn on the switch by pressing it, I see in the log:
```
debug 2022-04-14 16:18:19: Received Zigbee message from 'hallway_1_switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-04-14 16:18:19: MQTT publish: topic 'zigbee2mqtt/hallway_1_switch', payload '{"linkquality":105,"state":"OFF"}'
```

but if I attempt to set the switch by issuing a command in zigbee2mqtt:
```
debug 2022-04-14 16:18:45: Received MQTT message on 'zigbee2mqtt/hallway_1_switch/set' with data 'OFF'
debug 2022-04-14 16:18:45: Publishing 'set' 'state' to 'hallway_1_switch'
error 2022-04-14 16:19:12: Publish 'set' 'state' to 'hallway_1_switch' failed: 'Error: Command 0xa89f0111e00bfdb0/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
debug 2022-04-14 16:19:12: Error: Command 0xa89f0111e00bfdb0/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
info  2022-04-14 16:19:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'hallway_1_switch' failed: 'Error: Command 0xa89f0111e00bfdb0/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"hallway_1_switch"},"type":"zigbee_publish_error"}'
```
Attached are the logs for the day.
Power cycling of the switches occurred at 2022-04-14 15:55

Few commands were issued to set the switch that worked fine, and then at 2022-04-14 15:59:22 you can see the first error again.

All other switches work just fine, just those two that supports 2-way exhibit this behaviour.

If I reset the switch and re-pair them, they work for a little while until the problem occurs again.

I've upgraded to the latest firmware 20220219 without much difference.
What's bizarre is that I never had that issue before upgrading to 1.25.0, but even after restoring 1.24.1 from a backup, the same issue continued.

### What did you expect to happen?

Switches can be set on/off continuously without ever tailng

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220219

### Adapter

zzh!

### Debug log

Attached are the logs for the day.
Power cycling of the switches occurred at 2022-04-14 15:55
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8486733/log.txt)

johnnykang, 2022-04-14:
i believe we both have the same issue #12062

dirkjan2302, 2022-04-14:
Same here, reverted back to previous version.

johnnykang, 2022-04-14:
> Same here, reverted back to previous version.

What version did you revert to? Which device?

jyavenard, 2022-04-14:
> 

Did that fix it for you? Which version did you revert to?

tauzN, 2022-04-15:
Most likely also related to this: #12180

johnnykang, 2022-04-15:
> Most likely also related to this: #12180 
> 
> ```
> Debug Received Zigbee message from '0x0c4314fffe18a6f7', type 'commandDownClose', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID null
> ```

Unlikely. Not from the log i read. 

My(our) issue is the client device doesnt even respond to the command. And #12180 showes that the device did respond to command but no converter is found.

jyavenard, 2022-05-18:
Kevin from 3A has received firmware updates. I dropped by his shop and he updated the firmware in 5 minutes.

Seems to be working fine now, it's been 30 minutes and so far so good (it would have failed within 2 minutes before that).

So apparently, the switch received an invalid zigbee command that made it crashed, so the fix is to just ignore that command.

Now which command it is and why it started all of a sudden is unknown.

Is it zigbee2mqtt that started to send it from last month or is it a firmware update of my other zigbee devices I don't know

trentbraidner, 2022-05-18:
Thats awesome to hear @jyavenard 
Thanks for the update

Ill contact Kevin to discuss further

Mario-Bengtson, 2022-06-04:
I have been having exactly the same problem.
Did you guys get it solved? Is it a firmware update of the HGZB-43 switch or the zigbee coordinator?

johnnykang, 2022-06-04:
Kevin from 3A is gonna flash the devices (gang switches). I believe his team has got a solution but it requires a device(gang switches) firmware re-program. (not the zigbee  coordinator)

notanotherbrick1, 2022-06-04:
Thank you very much for the fast response! Do you have his contact information or know where can I find the new firmware to download?

johnnykang, 2022-06-05:
try this - https://3asmarthome.com/products on the top right corner

notanotherbrick1, 2022-06-05:
Thank you very much johnnykang! I really appreciate that! I wonder if I download the firmware of my old HGZB-43 that are working and upload in the 2 new HGZB-43 that are not working it would fix the problem... I would have to learn how to download and flash the old firmware, but it is something that may work...

best

closed 2022-07-14 by github-actions[bot]