Koenkk/zigbee2mqtt #12175: Scenes do not recall color temperature in IKEA L1529

merlinschumacher, 2022-04-14:
### What happened?

I've created  scenes where the color temperature of my IKEA L1529 is set to varying temperatures. Recalling the scenes from either MQTT or the web GUI does not change the temperature. The brightness and on/off settings are set properly when recalling a  scene. This happens for device scenes and group scenes. It works properly for other lamps in the same group, but not for this IKEA lamp. 
The GUI shows a change to the expected temperature after the recall, but the lamp won't physically change. This also fails, when the device is the only one in the group.
Setting the color directly from the GUI or via MQTT works fine.

### What did you expect to happen?

That the color temperature is set correctly for the device when recalling a scene.

### How to reproduce it (minimal and precise)

Recall a  scene that changes the temperature of a IKEA L1529.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220103

### Adapter

zzh!

### Debug log

[zigbeelog.txt](https://github.com/Koenkk/zigbee2mqtt/files/8491681/zigbeelog.txt)

sjorge, 2022-05-15:
How did you create the scene? with scene_add or scene_store?

merlinschumacher, 2022-05-20:
I've created the scenes via the web interface. I'm not sure which one it uses.

sjorge, 2022-05-20:
OK that uses the store_scene method ...

@Koenkk I guess the workaround for < ZCL 7 scene support where we translate color_temp into xy doesn't happen for scene_store, so I think there is no way to add color_temp in scenes using scene store on these ikea devices?

Koenkk, 2022-05-21:
@sjorge yes the workaround doesn't happen. With `scene_store` we just say restore to your current state whenever scene X is called. If the device doesn't adhere we can't do anything about that. A workaround would be to use the `scene_add` instead (since there z2m explicitly provides the state to restore to)

sjorge, 2022-05-21:
OK so exactly as I thought, switching to using `scene_add` is the solution here then.

closed 2022-06-28 by github-actions[bot]