Koenkk/zigbee2mqtt #12184: Water consumption working bad on RTX ZVG1

kriskristoferson, 2022-04-15:
### What happened?

[RTX ZVG1](https://www.zigbee2mqtt.io/devices/ZVG1.html#rtx-zvg1)
Water consumption is shown incorrectly.
I poured about 20 liters of water, the valve first showed 2.7, then 1.5. Readings come up every few minutes, but the measurement goes down instead of increasing, and it doesn't have to do with the amount of water actually passed.
The seller informed me that I am not the only one with the problem with zigbee2mqtt, the reading is correct on the original gates.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20211217

### Adapter

Sonoff CC2652P zStack3x0

kriskristoferson, 2022-04-16:
Additionally, errors from the joining process:
```
Warning zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #16 with data {"dp":16,"datatype":0,"data":{"type":"Buffer","data":[0]}}
Warning zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #17 with data {"dp":17,"datatype":0,"data":{"type":"Buffer","data":[0]}}
```

Koenkk, 2022-04-17:
See https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats on how to find out what these data points mean

toki-it, 2022-05-24:
Any updates? I have the same issue with water consumption. @kriskristoferson did you manage to resolve issue?

kriskristoferson, 2022-05-25:
Unfortunately, there is no progress.
My first ZVG1 is not counting well even in the Tuya app. I bought the second one and it works fine in the Tuya application, unfortunately not in the Z2M.
I debugged the valve according to @Koenkk  instructions. DP # 16 and # 17 are not visible in Tuya IOT but they are rather unimportant, the flow data is being reported, it looks a bit like no summation of readings is done.

ooii, 2022-06-22:
Hi,

I have the same issue. I have of these valves and all are reporting 0.00 l. If I can help, please tell me how.

Sesshoumaru-sama, 2022-07-03:
Some here. I have one of these vales and it always reports 0.00L consumed.

henworth, 2022-07-03:
I concur, both of mine are the same way. I also cannot change the built in timer, no matter what I set they always shut off at 10 minutes. Happy to open another issue if that is desired.

ytzelf, 2022-07-12:
Edited with correct link

fhempy, 2022-08-03:
Hi,
I sniffed the traffic on my valve and can confirm that some valves have no consumption reporting. Mine (cost me around 40€) reports 0 via DP5 and DP6. I checked the site I bought the valve from (https://www.pearl.at/at-a-ZX1193-3103.shtml) and it doesn't mention consumption reporting either.

You can try the external converter I built here:
https://github.com/Koenkk/zigbee2mqtt/issues/13199
It supports normal timers and weather delay. I'm going to implement irrigation cycle as well.

closed 2022-09-10 by github-actions[bot]

yah996, 2022-10-29:
HI, any progress ion this issue? Thanks

yah996, 2022-10-30:
If have traced the Tuna response on DP5. This values are correct reported during my tests.

{"result":{"devId":"bf3b1da36978dada99uczr","datas":[{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667115020633,"eventName":"water once","eventTimeStr":"2022-10-30 07:30:20:633","eventDetail":"4.30L","requestFrom":"device itself"},{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667115020497,"eventName":"water once","eventTimeStr":"2022-10-30 07:30:20:497","eventDetail":"100.00L","requestFrom":"device itself"},{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667113891661,"eventName":"water once","eventTimeStr":"2022-10-30 07:11:31:661","eventDetail":"15.70L","requestFrom":"device itself"},{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667113785285,"eventName":"water once","eventTimeStr":"2022-10-30 07:09:45:285","eventDetail":"8.20L","requestFrom":"device itself"},{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667113683052,"eventName":"water once","eventTimeStr":"2022-10-30 07:08:03:052","eventDetail":"0.00L","requestFrom":"device itself"}],"docType":1,"hasNext":false,"totalCount":0,"currentPageStartRowKey":"???"},"t":1667119219472,"success":true,"status":"ok"}

yah996, 2022-10-30:
> See https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats on how to find out what these data points mean

If have traced the Tuya (Hardware Gateway) response on DP5. This values are correct reported during my tests.

{"result":{"devId":"bf3b1da36978dada99uczr","datas":[{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667115020633,"eventName":"water once","eventTimeStr":"2022-10-30 07:30:20:633","eventDetail":"4.30L","requestFrom":"device itself"},{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667115020497,"eventName":"water once","eventTimeStr":"2022-10-30 07:30:20:497","eventDetail":"100.00L","requestFrom":"device itself"},{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667113891661,"eventName":"water once","eventTimeStr":"2022-10-30 07:11:31:661","eventDetail":"15.70L","requestFrom":"device itself"},{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667113785285,"eventName":"water once","eventTimeStr":"2022-10-30 07:09:45:285","eventDetail":"8.20L","requestFrom":"device itself"},{"devId":"bf3b1da36978dada99uczr","eventType":"Report","sourceDetail":"","eventTime":1667113683052,"eventName":"water once","eventTimeStr":"2022-10-30 07:08:03:052","eventDetail":"0.00L","requestFrom":"device itself"}],"docType":1,"hasNext":false,"totalCount":0,"currentPageStartRowKey":"???"},"t":1667119219472,"success":true,"status":"ok"}

yah996, 2022-10-30:
> See https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats on how to find out what these data points mean

please see above

fhempy, 2022-10-31:
Actually it should work already. What happens in your case when the valve is closed? Are values reported or do you get an error message in z2m?

yah996, 2022-10-31:
The valve itself works perfectly, On/Off. 

If the valve closed, **yes** I got an error (warning) message like  DP6 is undefined. Strange to me, because I con not see any DP6 on Tuya app debug. 

Again, only the watering usage is incorrect. The fromZigbee converter (zigbee2mqtt) includes in the ZVG1 section an strange converter entry. 

As you wrote:  // Assume value is reported in fl. oz., converter to litres
return {water_consumed: (value / 33.8140226).toFixed(2)};

Thanks.

yah996, 2022-10-31:
I assume you converted the value for you to see in US format, fl. oz. This can be confusing, because the Z2M show these values in Liter.

ScratMan, 2023-05-02:
Hello, I own two _TZE200_akjefhj5 valves (Saswell SAS980SWT-7-Z01 with water consumption reporting), and when pairing it with Zigbee2MQTT it is recognized as a RTX ZVG1 valve, but the water consumption is wrong. The ZVG1 converter considers the reported value as fl oz, while in the Tuya IOT app the status set indicates it's tenths of liters on my valves.

I made my own external converter I shared here : https://github.com/Koenkk/zigbee-herdsman-converters/issues/2795#issuecomment-1532210998
You may try it and see if it helps.