Koenkk/zigbee2mqtt #12188: TS011F_plug_3 switching state not fully responding

WBoerhoop, 2022-04-15:
### What happened?

Did a new pairing with the TS011F_plug_3, 6 Devices were neatly created. 
After creating it I want to turn on the device via state and that worked,  but the "state" was still off and no change in the icon.

### What did you expect to happen?

Changing the display to "on"  and the state icon with a red dot in it.
Also the option to turn the state off again which doesn't work.

### How to reproduce it (minimal and precise)

delete the device and paired it again.
[image]

### Zigbee2MQTT version

V.1.25.0-latest dev

### Adapter firmware version

20210120

### Adapter

Slaesh's CC2652RB

WBoerhoop, 2022-04-16:
[after restart.txt](https://github.com/Koenkk/zigbee2mqtt/files/8499536/after.restart.txt)

SargonofAssyria, 2022-04-17:
I had the same problem. In my case the problem was caused by the child_lock.
I created an external converter from the same device but without the child_lock.
Then everything worked as it should.

WBoerhoop, 2022-04-17:
> I had the same problem. In my case the problem was caused by the child_lock. I created an external converter from the same device but without the child_lock. Then everything worked as it should.

Ok, I can try that, will you upload your "external converter" ??

SargonofAssyria, 2022-04-17:
[TS011F_TZ3000.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/8501853/TS011F_TZ3000.js.txt)
Remove the .txt and place the file in the data dir and add the external converter.

WBoerhoop, 2022-04-17:
Used external converter file and Check, it´s working again, 
Thanks SargonofAssyria

closed 2022-04-17 by WBoerhoop