Koenkk/zigbee2mqtt #12189: Sengled E11-N1EA "Device didn't respond to OTA request"

dmwyatt, 2022-04-15:
### What happened?

OTA update does not work on this bulb. I have another bulb of the same model and it updated fine.

Relevant log snippet (complete log attached):

```
info  2022-04-15 16:16:56: MQTT publish: topic 'zigbee2mqtt/Red lamp bulb', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":81,"state":"OFF","update":{"state":"available"},"update_available":true}'
debug 2022-04-15 16:17:10: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Red lamp bulb","transaction":"uvtzy-1"}'
info  2022-04-15 16:17:10: Updating 'Red lamp bulb' to latest firmware
debug 2022-04-15 16:17:11: Received Zigbee message from 'Red lamp bulb', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2022-04-15 16:17:11: Updating to latest '0xb0ce18140004e9b6' (E11-N1EA)
debug 2022-04-15 16:17:11: Using endpoint '1'
debug 2022-04-15 16:17:20: Received Zigbee message from 'Silver Lamp Bulb 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,199467]}' from endpoint 1 with groupID 0
info  2022-04-15 16:17:20: MQTT publish: topic 'zigbee2mqtt/Silver Lamp Bulb 1', payload '{"brightness":254,"energy":19.95,"linkquality":87,"power":0,"state":"OFF","update":{"state":"idle"},"update_available":false}'
debug 2022-04-15 16:17:36: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2022-04-15 16:17:41: Update of 'Red lamp bulb' failed (Error: Device didn't respond to OTA request)
info  2022-04-15 16:17:41: MQTT publish: topic 'zigbee2mqtt/Red lamp bulb', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":81,"state":"OFF","update":{"state":"available"},"update_available":true}'
info  2022-04-15 16:17:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Red lamp bulb"},"error":"Update of 'Red lamp bulb' failed (Device didn't respond to OTA request)","status":"error","transaction":"uvtzy-1"}'
error 2022-04-15 16:17:41: Update of 'Red lamp bulb' failed (Device didn't respond to OTA request)
debug 2022-04-15 16:17:41: Error: Device didn't respond to OTA request
    at requestOTA (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15)
    at Object.updateToLatest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:283:21)
    at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:256:21)
```

### What did you expect to happen?

The OTA to work.

### How to reproduce it (minimal and precise)

No idea!  

### Zigbee2MQTT version

1.25.0-dev commit: 84c8da25

### Adapter firmware version

20210120

### Adapter

Slaesh's CC2652RB stick

### Debug log

[log-2022-04-15.15-37-33.log](https://github.com/Koenkk/zigbee2mqtt/files/8498363/log-2022-04-15.15-37-33.log)

S3cBar0n, 2022-04-17:
I get the same exact error with several of these bulbs.

dmwyatt, 2022-05-18:
Still s problem.

padadof2, 2022-06-15:
I'm having the same problem as well.  I have 5 of the sengled E21-N1EA, and they give the message "(No image available for imageType '32972')", but my two E11-N1EA, give me a message of "(Device didn't respond to OTA request)"  These are the only devices I have on my zigbee network that do OTA's

TimDowker, 2022-06-29:
Same issue but with E11-G13 bulbs.  Get "(Device didn't respond to OTA request)".

TimDowker, 2022-07-07:
Still having this issue even after the latest update.  Confirmed it is still an issue in version 1.26.0-1.  

Just to reiterate, I get this message when I try to check for updates on any of my Sengled E11-G13 bulbs - "Device didn't respond to OTA request".  

I have tried restarting them immediately prior to checking for an update.  Same results.

dmwyatt, 2022-08-07:
Well it's still a problem, so bad bot.

fiatguy85, 2022-08-14:
Yes, I'm a new user, also having this issue with Sengled E11-N1EA.

dmwyatt, 2022-09-14:
Not stale

amsmith525, 2022-09-30:
I too am having this issue - only on one of the bulbs though. One of them updated for around 10 minutes. The other is having this error

Koenkk, 2022-09-30:
There is probably little we can do from the z2m side. However it might help to temporarily move the bulb very close to the coordinator, repair it and try again.

dmwyatt, 2022-10-31:
Not stale

amsmith525, 2022-11-18:
I was able to update all bulbs after updating my ZigBee stick to the latest coordinator firmware per this page -- https://www.zigbee2mqtt.io/guide/adapters/

closed 2022-12-27 by github-actions[bot]

villasenor, 2023-05-21:
@Koenkk - what is your theory on why this is happening? If this won't work with z2m, could I get a Sengled Hub, update the bulbs with that, and then re-pair them back to z2m? If that is true, it seems odd that the Sengled Hub could update the bulbs but not z2m?

Koenkk, 2023-05-22:
> If this won't work with z2m, could I get a Sengled Hub, update the bulbs with that, and then re-pair them back to z2m?

Yes this will work

> If that is true, it seems odd that the Sengled Hub could update the bulbs but not z2m?

If true, please provide a sniff of both updating via hub and z2m so we can check the diff. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

fireheadman, 2023-11-05:
anything new on this issue?   I have 2 bulbs that show updates available, they just spin forever trying to get from v21 --> v30.   Nothing ever updates.    Half tempted to also get the hub from AMZ, update them and return it.    Wish zigbee2mqtt would follow through.