Koenkk/zigbee2mqtt #12198: Not receiving ON state changes from HK-DIM-A

cmsj, 2022-04-17:
### What happened?

I've installed two Candeo HK-DIM-A switches in different parts of my house and I've noticed that while I get events for changing the brightness and pressing the button to turn the lights off, I almost never get events when I press the button to turn the lights back on.

Here is an example of the logging I see for changing the brightness and pressing the button to turn the lights off. I then immediately pressed the button to turn the lights back on and no event was logged for it.
```
Zigbee2MQTT:info  2022-04-17 15:37:33: MQTT publish: topic 'zigbee2mqtt/Dining Room Spots', payload '{"brightness":246,"linkquality":29,"state":"ON"}'
Zigbee2MQTT:info  2022-04-17 15:37:34: MQTT publish: topic 'zigbee2mqtt/Dining Room Spots', payload '{"brightness":239,"linkquality":29,"state":"ON"}'
Zigbee2MQTT:info  2022-04-17 15:37:36: MQTT publish: topic 'zigbee2mqtt/Dining Room Spots', payload '{"brightness":246,"linkquality":21,"state":"ON"}'
Zigbee2MQTT:info  2022-04-17 15:37:37: MQTT publish: topic 'zigbee2mqtt/Dining Room Spots', payload '{"brightness":254,"linkquality":25,"state":"ON"}'
Zigbee2MQTT:info  2022-04-17 15:37:39: MQTT publish: topic 'zigbee2mqtt/Dining Room Spots', payload '{"brightness":254,"linkquality":43,"state":"OFF"}'
```

I'm quite new to Zigbee2MQTT, so I don't know if this is a generic limitation, or something specific to these switches. I saw that @kangaroomadman added support for these devices about 6 months ago, and there was a subsequent change to limit how frequently they send brightness updates - perhaps that is related?

### Zigbee2MQTT version

1.25.0 commit: 6f1460e

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

closed 2022-05-25 by github-actions[bot]

cmsj, 2022-06-04:
If I turn the switch on physically, then tell zigbee2mqtt's "Exposes" view of the device to refresh the state of the on/off button, it still doesn't seem to understand that the light is on:

```
Jun 04 19:38:36 gnubert 15dba2cdacaf[185186]: Zigbee2MQTT:debug 2022-06-04 19:38:36: Received MQTT message on 'zigbee2mqtt/Bedroom Dimmer Switch/get' with data '{"state":""}'
Jun 04 19:38:36 gnubert 15dba2cdacaf[185186]: Zigbee2MQTT:debug 2022-06-04 19:38:36: Publishing get 'get' 'state' to 'Bedroom Dimmer Switch'
Jun 04 19:38:36 gnubert 15dba2cdacaf[185186]: Zigbee2MQTT:debug 2022-06-04 19:38:36: Received Zigbee message from 'Bedroom Dimmer Switch', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Jun 04 19:38:36 gnubert 15dba2cdacaf[185186]: Zigbee2MQTT:info  2022-06-04 19:38:36: MQTT publish: topic 'zigbee2mqtt/Bedroom Dimmer Switch', payload '{"brightness":140,"linkquality":163,"state":"OFF"}'
```

kangaroomadman, 2022-06-05:
I’ve not had a problem with my Candeo dimmer- it’s never missed an on/off/brightness command. 

The brightness update change was to reduce the amount of messages sent as, in my network, so many messages were arriving at the coordinator out of order which gives strange behaviour when relaying the brightness values to lights. 

I can only guess that there may be something with your Zigbee network or perhaps the firmware version on the Candeo dimmer? Are you able to set up a separate network with just the dimmer on? Does it still miss ‘on’ commands?

cmsj, 2022-07-01:
I spoke with Candeo, who initially assumed it was my Zigbee network/hub, but have since contacted me to say there was indeed a firmware issue for a short run of production and are going to RMA my switches. Hopefully that will be the end of it, I just need to get around to swapping back to the dumb switches!
Thanks for the reply :)