Koenkk/zigbee2mqtt #12202: [New device support]: LIDL Livarno Home ceiling light

Mecallie, 2022-04-17:
### Link

https://www.lidl.nl/p/livarno-home-led-plafondlamp-zigbee-smart-home/p100335196

### Database entry

Device '0x2c1165fffe6d6817' with Zigbee model 'TS0504B' and manufacturer name '_TZ3210_sroezl0s'

### Comments

I wanted to add this RGBWW Zigbee Light from LIDL, but it is not recognized:
Device '0x2c1165fffe6d6817' with Zigbee model 'TS0504B' and manufacturer name '_TZ3210_sroezl0s' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

### Supported color modes

RGB+WW

Koenkk, 2022-04-18:
`_TZ3210_sroezl0s` is already supported in the dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html). Next release will be on 1 May

closed 2022-04-18 by Koenkk

Mecallie, 2022-04-19:
Great! Just tested it: on/off works. Changing the colour works, but seems very low brightness even at max (might be correct).
The white-tint does not work: changing the slider does not do anything. Also the lowest setting seems to be very bright, especially when compared to the colour mode which is very dim even at 100% brightness.

Feels a bit like something is not being used? Is it possible there are two white channels available and only one is being used? (RGB-WW?)

luchtgitaar, 2022-04-25:
> Great! Just tested it: on/off works. Changing the colour works, but seems very low brightness even at max (might be correct). The white-tint does not work: changing the slider does not do anything. [...]

Are you still experiencing this issue? I am interested in this light.

Mecallie, 2022-05-02:
> > Great! Just tested it: on/off works. Changing the colour works, but seems very low brightness even at max (might be correct). The white-tint does not work: changing the slider does not do anything. Also the lowest setting seems to be very bright, especially when compared to the colour mode which is very dim even at 100% brightness.
> > Feels a bit like something is not being used? Is it possible there are two white channels available and only one is being used? (RGB-WW?)
> 
> Are you still experiencing this issue? I am interested in this light.
> 
> Edit: maybe you have to enable white channel by sending a payload. See: https://www.zigbee2mqtt.io/devices/TS0504B.html

The issue remains. All that is working correctly for now is dimming it when it is set to white (and then it is still quite bright) and changing the colours (but very dim). Changing the white tint does nothing.

As far as I understand it: sending the payload is exactly what z2m should do when a device is supported correctly?

Timminater, 2022-07-08:
Is RGB+CCT working for this light now?
I'm interested in ordering one, but mainly want to use it for Cold to Warm White light.