Koenkk/zigbee2mqtt #12204: SMHM-I1 fix battery level and voltage

unpack8866, 2022-04-18:
### What happened?

Repair battery level and voltage; But I don't know how to modify the voltage unit.
Voltage 2800v is modified to 2800mv; How to modify?

[image]

const device = {
zigbeeModel: ['SMHM-I1', 'PIR_TPV12'],
model: 'SMHM-I1',
vendor: 'HEIMAN',
description: 'Smart motion sensor',
fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery],
toZigbee: [],
exposes: [e.occupancy(), e.battery_low(), e.battery(), e.voltage()],
meta: {battery: {voltageToPercentage: '3V_2500'}},
configure: async (device, coordinatorEndpoint, logger) => {
const endpoint = device.getEndpoint(1);
const bindClusters = ['genPowerCfg'];
await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
await reporting.batteryPercentageRemaining(endpoint);
await reporting.batteryVoltage(endpoint);
},
};

module.exports = device;

### Zigbee2MQTT version

1.25

### Adapter firmware version

20220301

### Adapter

cc2652p

Koenkk, 2022-04-18:
Change `e.voltage()` -> `e.battery_voltage()`

unpack8866, 2022-04-19:
> Change `e.voltage()` -> `e.battery_voltage()`

It's right! Please fix it, thank you!
[image]

const device = {
    zigbeeModel: ['SMHM-I1', 'PIR_TPV12'],
    model: 'SMHM-I1',
    vendor: 'HEIMAN',
    description: 'Smart motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.battery(), e.battery_voltage()],  
    meta: {battery: {voltageToPercentage: '3V_2500'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const bindClusters = ['genPowerCfg'];
        await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.batteryVoltage(endpoint);

   },
};

module.exports = device;

Koenkk, 2022-04-19:
great, added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-19 by Koenkk