Koenkk/zigbee2mqtt #12206: Aqara Water leak sensor SJCGQ11LM after pairing disconnects from coordinator after some time

acaonweb, 2022-04-18:
### What happened?

Pairing the device is without any problem, and it works good.
But after some minutes it looses the connection with coordinator and the only way is to repair it again.

### What did you expect to happen?

The device paired to Xiaomi Gateway isn't affected by this problem

### How to reproduce it (minimal and precise)

Pair device, check zigbee network map after some time to verify disconnection

### Zigbee2MQTT version

1.25.0 

### Adapter firmware version

20210120

### Adapter

zzh! CC2652R Multiprotocol RF Stick

Koenkk, 2022-04-18:
Does the SJCGQ11LM still send data updates? If it just doesn't appear connected on the network map, this is fine, see: https://www.zigbee2mqtt.io/guide/faq/#why-are-some-links-missing-from-my-networkmap

acaonweb, 2022-04-19:
> Does the SJCGQ11LM still send data updates? If it just doesn't appear connected on the network map, this is fine, see: https://www.zigbee2mqtt.io/guide/faq/#why-are-some-links-missing-from-my-networkmap

I'm pretty sure it doens't update its status, but let me have a deep check

McGiverGim, 2022-04-19:
@acaonweb press the top of the device for one instant (you will see a blue light). The device will send a message to the coordinator with the current status.

acaonweb, 2022-04-20:
After a dozen of attempt, I confirm that although pairing is successful, and the status is sent when is connected to coordinator, after some time the device loose the network and don't send status in anyway, neither with water, nor by pressing it. The only way is repair again. The device works good when linked to xiaomi gateway

Koenkk, 2022-04-21:
@acaonweb to debug that further I will need to have a sniff: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Provide a sniff starting at the point where you pair the device until it doesn't send any updates anymore, after this press the button on the device and stop sniffing.

acaonweb, 2022-04-21:
Looking at the link you provided, I understand I need to flash my slaesh zzh dongle with a different firmware, right?
I have only one and a lot of automations in domoticz. 
I would avoid compromising my system, there's another way?

Koenkk, 2022-04-22:
@acaonweb you need an additional dongle and have to flash it with a sniffer firmware.

acaonweb, 2022-04-22:
Oh, well maybe I'll prefer send you aqara device 😄😄

Koenkk, 2022-04-22:
@acaonweb I already have this device and it works fine in my case, so it is probably something in your network.

acaonweb, 2022-04-22:
Ok... I'll plan it 
Thanks

closed 2022-05-30 by github-actions[bot]