Koenkk/zigbee2mqtt #12208: Zigbee2MQTT fails to start after adding log_output paramet

OlsiIgitt, 2022-04-18:
### What happened?

When I add the line 'log_output: file' (dots -> spaces)

>advanced:
>..log_output: file

then Zigbee2MQTT will always crash after starting with following information in cmd window:

>npm ERR! code ELIFECYCLE
>npm ERR! errno 1
>npm ERR! zigbee2mqtt@1.25.0 start: `node index.js`
>npm ERR! Exit status 1
>npm ERR!
>npm ERR! Failed at the zigbee2mqtt@1.25.0 start script.
>npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

Z2M is running under Windows 10 !

### What did you expect to happen?

All information should be logged (inclusively debug information) into log file  without crash.

### How to reproduce it (minimal and precise)

add following lines to configuration.yaml:

advanced:
  log_output: file

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219

### Adapter

Zigbee 3.0 USB Dongle Plus

### Debug log

Nothing -> no log file created

lux73, 2022-04-18:
log_output? try these commends below - works for me

advanced:
..log_directory: /var/log/zigbee2mqtt
..log_file: zigbee2mqtt.log

lux73, 2022-04-18:
please try `log_file`

OlsiIgitt, 2022-04-18:
I tried this:

advanced:
..log_level: debug
..log_directory: c:\progs\z2m\data\test
..log_file: test.log
.. #log_output: test.log

This will work, but activating 'log_output' makes Z2M crash with 100 % certainty (Windows); indepent of paramters I try behind "log_output: ".

> log_output is not the correct command -> https://www.zigbee2mqtt.io/guide/configuration/logging.html#logging

There I can find the statement:

>'# Optional: Output location of the log (default: shown below), leave empty to supress logging (log_output: [])
>'# possible options: 'console', 'file', 'syslog'
>'  log_output:
>'    - console
>'    - file

What does it mean???

Koenkk, 2022-04-18:
log output should be an array, see the docs: https://www.zigbee2mqtt.io/guide/configuration/logging.html, so e.g. `log_output: ["console", "file"]`

OlsiIgitt, 2022-04-19:
Thanks, Koen.

'  log_output: [file]'
or
'  log_output: [file, console]'
etc. works fine.

Sorry, the reason for the problem was my poor knowledge of JavaScript. Honestly the only thing I know about JS is, that it exists.
Possibly the word "array" in the description could have helped. Or another example like above besides '  log_output: []'?

But what is not a JS thing:
When I make mistakes in defining '  log_directory: ...' (or others), I will get an comprehensive information from Z2M about that. Doing mistakes with '  log_output: ...' Z2M punishes me with absolutely no message, no matter if '  log_level: debug' or not.

Bye...

OlsiIgitt, 2022-04-20:
You are absolutely right.
I apologize for that.
I only had eyes for the red marked error messages...

I vow to do better.