Koenkk/zigbee2mqtt #12212: Failed to configure Develco SMSZB-120

Bascht74, 2022-04-18:
### What happened?

Hi, I joined 3 Develco SMSZB-120 and after that I get some errors from time to time: 

```
Zigbee2MQTT:error 2022-04-19 00:13:04: Failed to configure 'Rauchmelder_EG_Flur', attempt 1 (Error: Bind 0x0015bc003100c740/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-04-19 00:13:26: Failed to configure 'Rauchmelder_Aussen_Carport', attempt 1 (Error: Bind 0x0015bc003100bb13/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-04-19 00:15:22: Failed to configure 'Rauchmelder_Aussen_Schuppen', attempt 1 (Error: Bind 0x0015bc003100bea0/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-04-19 00:15:52: Failed to configure 'Rauchmelder_OG_Flur', attempt 1 (Error: Bind 0x0015bc003100bae5/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-04-19 00:18:34: Failed to configure 'Rauchmelder_EG_Flur', attempt 2 (Error: Bind 0x0015bc003100c740/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-04-19 00:18:35: Failed to configure 'Rauchmelder_Aussen_Carport', attempt 2 (Error: Bind 0x0015bc003100bb13/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-04-19 00:20:29: Failed to configure 'Rauchmelder_Aussen_Schuppen', attempt 2 (Error: Bind 0x0015bc003100bea0/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-04-19 00:20:59: Failed to configure 'Rauchmelder_OG_Flur', attempt 2 (Error: Bind 0x0015bc003100bae5/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
```

I tried to re-join them, but this error still remains. 

Is there anything I can do about that?

### What did you expect to happen?

That Z2M can configure the device and the error is gone

### How to reproduce it (minimal and precise)

1. Join the devices
2. Find the error from time to time in the logs.

### Zigbee2MQTT version

Zigbee2MQTT Version 1.25.0 commit: unknown, Frontend Version 0.6.80

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

How to enable debug right now? It seems to be changed after the newest add-on for Home Assistant.
I have two options:
zigbee_shepherd_devices
zigbee_herdsman_debug

The log level can't be changed via the addon (but that would make sence). 
I need some information how to archive this...

smulle48, 2022-04-22:
The only thing I can find about that error, is that it apparently is something with the Conbee2 stick

https://github.com/Koenkk/zigbee2mqtt/issues/5223

To enable debug log, just change the last line to true :-)

[image]

Bascht74, 2022-04-24:
Hi, 

here is a log with `debug: true`:

https://pastebin.com/j7c1m9HL

Rauchmelder_Aussen_Schuppen --> is one example of the three devices...

```
2022-04-24T20:57:09.041Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 124
2022-04-24T20:57:09.961Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2022-04-24T20:57:09.961Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-04-24T20:57:09.965Z zigbee-herdsman:deconz:driver query aps data confirm
2022-04-24T20:57:09.971Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 125
2022-04-24T20:57:09.973Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xa5a8 request id: 11 confirm status: 0
2022-04-24T20:57:09.973Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-04-24T20:57:10.065Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-04-24T20:57:10.065Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-04-24T20:57:10.066Z zigbee-herdsman:deconz:driver query aps data indication
2022-04-24T20:57:10.072Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 126
2022-04-24T20:57:10.074Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 126 srcAddr: 0xa5a8 destAddr: 0x0 profile id: 0x0 cluster id: 0x8021 lqi: 255
2022-04-24T20:57:10.074Z zigbee-herdsman:deconz:frameParser response payload: 11,140
2022-04-24T20:57:10.074Z zigbee-herdsman:deconz:adapter resolve data request without a transSeq Nr.
2022-04-24T20:57:10.075Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-04-24T20:57:10.075Z zigbee-herdsman:deconz:adapter BIND RESPONSE - addr: 0xa5a8 status: 140
2022-04-24T20:57:10.075Z zigbee-herdsman:deconz:adapter BIND FAILED - addr: 0xa5a8 Error: status: 140
2022-04-24T20:57:10.076Z zigbee-herdsman:controller:endpoint Bind 0x0015bc003100bea0/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
Zigbee2MQTT:error 2022-04-24 22:57:10: Failed to configure 'Rauchmelder_Aussen_Schuppen', attempt 1 (Error: Bind 0x0015bc003100bea0/35 genBasic from '0x00212effff0657d6/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:167:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
2022-04-24T20:57:18.135Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 127
2022-04-24T20:57:18.137Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-04-24T20:57:18.137Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-04-24T20:57:23.117Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-04-24T20:57:23.117Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-04-24T20:57:23.122Z zigbee-herdsman:deconz:driver query aps data indication
2022-04-24T20:57:23.127Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 128
2022-04-24T20:57:23.129Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 128 srcAddr: 0xd83d destAddr: 0x0 profile id: 0x104 cluster id: 0x406 lqi: 255
2022-04-24T20:57:23.129Z zigbee-herdsman:deconz:frameParser response payload: 8,97,10,0,0,24,0
2022-04-24T20:57:23.130Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":97,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":24,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":55357,"endpoint":2,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2022-04-24T20:57:23.132Z zigbee-herdsman:controller:endpoint DefaultResponse 0x001788010649e511/2 1030(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-04-24T20:57:23.132Z zigbee-herdsman:deconz:adapter no response expected
2022-04-24T20:57:23.132Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

Bascht74, 2022-04-24:
Status of the device:
```
{
    "battery": 100,
    "battery_low": false,
    "linkquality": 240,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "test": false,
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "voltage": 3000,
    "alarm": null,
    "fault": null,
    "max_duration": null,
    "reliability": null,
    "temperature": null
}
```

[image]

[image]

[image]

[image]

closed 2022-06-01 by github-actions[bot]

RedJohn14, 2022-08-07:
Hi, I have the same error with the exact same device. Any news on this? :-(

akreker, 2022-09-01:
Hi, I also have the same error. But I have one device working. Don't know why. I thought it is because of the newer firmware of the device. But there is another one with the older firmware.
The temperature is also only working on the "working" device.

I have a Conbee II with newest firmware.

[image]

admlko, 2022-12-17:
Same situation, one device shows firmware 4.0.4, others don't show anything.
Also Conbee II with latest firmware.

AKA88NL, 2022-12-20:
I have the same issue, 1 works, I have another 3 that do not work. Any way to troubleshoot this?

The-Mech4nic, 2023-01-03:
I've just started working with 6 of those, and discovered several issues while using Zigbee2MQTT. After updating to the latest firmware (4.0.5), I quickly found out his causes issues with Z2M integration; the ALARM: "OFF" payload would now actually enable the alarm. #15795 

After rolling back the firmware (4.0.2), this seems to be working again. However, the 6 devices are still not initialized in the same way! Two out of 6 correctly reported temperature and communicated a value for Fault and Reliability. The other 4 did not. After copying the "Reporting" configuration from the functioning devices, now all 6 reported the temperature properly. If you copy this the temperature should also be reported:
[image]

Unfortunately the State template of the devices still differs between them, even though they are purchased at the same time and running the same firmware version (4.0.2). The sirens can be disabled and enabled, but most do not communicate Reliability and Fault, and all of them do not send a Zigbee payload when the physiscal Test button is pushed. I'm not sure how to fix this.

I'm also using the Conbee II. 

One of the alarms was going off randomly (sitting next to the others), without reporting Smoke. I tried re-pairing it, after which I also started to receive these same "Failed to configure" error messages in the log for some of the remaining alarms (not all...).

TheStigh, 2023-01-04:
> I've just started working with 6 of those, and discovered several issues while using Zigbee2MQTT. After updating to the latest firmware (4.0.5), I quickly found out his causes issues with Z2M [...]

ALARM: "OFF": Update Z2M to latest (1.29.0) version. That bug has been clearly discussed and fixed.
Conbee II:  Users seem to have issues that no other coordinator types have.
If you read the documentation and specs, a test (which ALARM: "ON" are), will not trigger and payload.

The-Mech4nic, 2023-01-04:
> ALARM: "OFF": Update Z2M to latest (1.29.0) version. That bug has been clearly discussed and fixed. Conbee II: Users seem to have issues that no other coordinator types have. If you read the documentation and specs, a test (which ALARM: "ON" are), will not trigger and payload.

Thanks for your reply. I'm aware this issue has been addressed, have been following the issue. I was waiting for the januari release according to your post, thanks for reminding me 1.29.0 is available already ;) After updating Z2M still receiving these configuration errors, must indeed be related to my Conbee II... 

> 2023-01-04 11:59:42Failed to configure 'Smoke alarm 2', attempt 3 (Error: Bind 0x0015bc0031016c88/35 genBinaryInput from '0x00212effff08579e/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:436:13) at Configure.configure (/app/lib/extension/configure.ts:117:13))

TheStigh, 2023-01-04:
I cannot conclude with Conbee being the "problem" but I've read posts of Conbee II owners adressing same issues that has been resolved by using a different coordinator. Thuogh, I also know that Conbee II users got this product to work.

Unfortunately I cannot help with Conbee II. I dont have such adapter nor do I plan to purchase one. When I make my integrations I always test with two different coordinators, zzh and Sonoff 3.0.

akreker, 2023-01-07:
It works! 

> I cannot conclude with Conbee being the "problem" but I've read posts of Conbee II owners adressing same issues that has been resolved by using a different coordinator. Thuogh, I also know that Conbee II users got this product to work.

I switched from Conbee II to  SONOFF ZigBee 3.0 USB Dongle Plus (ZBDongle-P). And now it works like a charm. Everything. With the CC2652 bootloader I had to move my z2m setup to a Pi, with Conbee I used ser2net.