Koenkk/zigbee2mqtt #12218: zigbee2mqtt: Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyUSB0'

deltabert, 2022-04-19:
### What happened?

Fresh install of Zigbee2MQTT according to these instructions: https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing
My coordinator is a Sonoff Zigbee 3.0 USB Dongle Plus.
It is found as /dev/ttyUSB0

My /opt/zigbee2mqtt/configuration.yaml:
`homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.178.222
  user: hass
  password: <My MQTT Pw>
  client_id: MY_CLIENT_ID
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: /dev/ttyUSB0
frontend:
  port: 8080
  host: 192.168.178.222`

The last log.txt file is:
`root@rpi4-had1:~# cat /opt/zigbee2mqtt/data/log/2022-04-19.15-41-21/log.txt
info  2022-04-19 15:41:23: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-04-19.15-41-21' filename: log.txt
info  2022-04-19 15:41:23: Starting Zigbee2MQTT version 1.25.0 (commit #6f1460e4)
info  2022-04-19 15:41:23: Starting zigbee-herdsman (0.14.20)
error 2022-04-19 15:41:24: Error while starting zigbee-herdsman
error 2022-04-19 15:41:24: Failed to start zigbee
error 2022-04-19 15:41:24: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2022-04-19 15:41:24: Exiting...
error 2022-04-19 15:41:24: Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyUSB0'
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12 `

I already performed the actions at: https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html
METHOD 2. and rebooted the RPi4, but lo success.

Any one an idea? 
This RPi4 is my test machine. I have also a production HA on NUCi5, and there zigbee2mqt is running fine.

### Zigbee2MQTT version

Latest version; downloaded and installed today

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

info  2022-04-19 16:16:52: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-04-19.16-16-50' filename: log.txt
info  2022-04-19 16:16:52: Starting Zigbee2MQTT version 1.25.0 (commit #6f1460e4)
info  2022-04-19 16:16:52: Starting zigbee-herdsman (0.14.20)
error 2022-04-19 16:16:53: Error while starting zigbee-herdsman
error 2022-04-19 16:16:53: Failed to start zigbee
error 2022-04-19 16:16:53: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2022-04-19 16:16:53: Exiting...
error 2022-04-19 16:16:53: Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyUSB0'
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12

deltabert, 2022-04-19:
BTW, I flashed the Sonoff Zigbee 3.0 USB Dongle Plus today with the lastest fw: 20220219

deltabert, 2022-04-19:
` sudo chown iotstack /dev/ttyUSB0
$ test -w /dev/ttyUSB0 && echo success || echo failure
success
`
OK, that is working.
Si i start zigbee2mqtt again:
`sudo systemctl start zigbee2mqtt.service
 sudo systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-04-19 19:54:27 CEST; 46s ago
   Main PID: 85919 (npm start)
      Tasks: 23 (limit: 4434)
     Memory: 133.3M
        CPU: 7.592s
     CGroup: /system.slice/zigbee2mqtt.service
             ├─85919 npm start
             ├─85930 sh -c node index.js
             └─85931 node index.js

Apr 19 19:54:27 rpi4-had1 systemd[1]: Started zigbee2mqtt.
Apr 19 19:54:28 rpi4-had1 npm[85919]: > zigbee2mqtt@1.25.0 start
Apr 19 19:54:28 rpi4-had1 npm[85919]: > node index.js
Apr 19 19:54:33 rpi4-had1 npm[85931]: Zigbee2MQTT:info  2022-04-19 19:54:33: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-04-19.19-54-30' >
Apr 19 19:54:33 rpi4-had1 npm[85931]: Zigbee2MQTT:info  2022-04-19 19:54:33: Starting Zigbee2MQTT version 1.25.0 (commit #6f1460e4)
Apr 19 19:54:33 rpi4-had1 npm[85931]: Zigbee2MQTT:info  2022-04-19 19:54:33: Starting zigbee-herdsman (0.14.20)
`
And the log file:
`ls -al /opt/zigbee2mqtt/data/log/2022-04-19.19-58-06
total 12
drwxr-xr-x  2 iotstack iotstack 4096 Apr 19 19:58 .
drwxr-xr-x 12 iotstack iotstack 4096 Apr 19 19:58 ..
-rw-r--r--  1 iotstack iotstack  271 Apr 19 19:58 log.txt
iotstack@rpi4-had1:~$ cat /opt/zigbee2mqtt/data/log/2022-04-19.19-58-06/log.txt
info  2022-04-19 19:58:08: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-04-19.19-58-06' filename: log.txt
info  2022-04-19 19:58:08: Starting Zigbee2MQTT version 1.25.0 (commit #6f1460e4)
info  2022-04-19 19:58:08: Starting zigbee-herdsman (0.14.20)
error 2022-04-19 19:59:14: Error while starting zigbee-herdsman
error 2022-04-19 19:59:14: Failed to start zigbee
error 2022-04-19 19:59:14: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2022-04-19 19:59:14: Exiting...
error 2022-04-19 19:59:14: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:100:27)
    at start (/opt/zigbee2mqtt/index.js:101:5)`

Any help please?

Eggn1n3, 2022-04-20:
I had the same issue and I solved it by unplugging the Sonoff, removing database.db and state.json files, inserting the Sonoff and starting my zigbee2mqtt again.

Eggn1n3, 2022-04-26:
Don't you have a 'data' dir?
[image]

closed 2022-06-04 by github-actions[bot]

eoinjones, 2022-10-01:
I had a similar issue and solved it by adding my user to the [dialout group](https://askubuntu.com/questions/58119/changing-permissions-on-serial-port).

wilwizard1979, 2022-11-11:
> I had a similar issue and solved it by adding my user to the [dialout group](https://askubuntu.com/questions/58119/changing-permissions-on-serial-port).

eoinjones, I'm having in Home Assistant the same reported issue above (Access Denied). Would it be possible to have the same solution of adding the user to the dialout group in Home Assistant? Any suggestion on how to do this since Home Assistant doesn't have Linux command line?