Koenkk/zigbee2mqtt #12220: Neo NAS-AB02B0 - msg.payload

RitaSghe, 2022-04-19:
### What happened?

Hi,
I write this function:

`msg.payload = {"alarm": true,
               "melody": 3,
               "duration": 10,
               "volume": "low"
              };

msg.topic = "zigbee2mqtt/SirenaCamera/set";
return msg;
`

and the siren start, but if I change the other parameters, nothing change ...
Where is the problem?

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

0x26720700 - 0x00212effff07359e - 0.6.71

### Adapter

Conbee II

RitaSghe, 2022-04-19:
Ok, the problem is I cannot send the parameters all together but only one at a time .....

closed 2022-05-27 by github-actions[bot]

Dreyk007, 2023-11-15:
Same issue for me

Chouhada, 2024-05-12:
I can confirm this is still a problem. Sending the parameters as separate payloads will work, but sending a single payload with multiple valid parameters always fails.
No idea if it's the alarm itself or something elsewhere. Happy to help debug if any developer has questions and can provide instructions.