Koenkk/zigbee2mqtt #12225: No converter available for NAS-AB02B0 (and NAS-PD07) with cluster manuSpecificTuya and type commandMcuSyncTime and data {payloadSize:XXX}

AlexKay88, 2022-04-20:
### What happened?

Debug log shows that Neo devices NAS-AB02B0 and NAS-PD07 report `commandMcuSyncTime`, but there ist no converter available.
(shown log messages are repeated several times)
Is it wanted to write an converter for these messages?

Device https://www.zigbee2mqtt.io/devices/NAS-AB02B0.html
`No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":59136}'`
Device https://www.zigbee2mqtt.io/devices/NAS-PD07.html
`No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":21513}'`

### Zigbee2MQTT version

1.25.0-dev

### Adapter firmware version

0x26720700

### Adapter

ConBee2

AlexKay88, 2022-04-20:
Or is it preferred to use existing  `ignore_tuya_set_time` in `fromZigbee.js` :
```
    ignore_tuya_set_time: {
        cluster: 'manuSpecificTuya',
        type: ['commandMcuSyncTime'],
        convert: (model, msg, publish, options, meta) => null,
    },

```
by adding
`fromZigbee: [ (...), fz.ignore_tuya_set_time],`
to Neo devices NAS-AB02B0 and NAS-PD07?

Koenkk, 2022-04-20:
Yes this can be ignored, please create a PR

AlexKay88, 2022-04-21:
I will test on my local system first and prepare a PR afterwards.

AlexKay88, 2022-04-25:
Are you sure, that it is not a conflict to add `fz.ignore_tuya_set_time`?

    ignore_tuya_set_time: {
        cluster: 'manuSpecificTuya',
        type: ['commandMcuSyncTime'],
        convert: (model, msg, publish, options, meta) => null,
    },

`devices/neo.js/NAS-AB02B0` and `NAS-PD07` contain `onEventSetLocalTime` / `onEventSetTime`
```
fromZigbee: [fz.neo_t_h_alarm, fz.ignore_basic_report, fz.ignore_tuya_set_time],
[...]
onEvent: tuya.onEventSetLocalTime,
```
or
```
fromZigbee: [fz.neo_nas_pd07, fz.ignore_tuya_set_time],
[...]
onEvent: tuya.onEventSetTime,
```
which uses `lib/tuya.js/onEventSetTime`:
```
async function onEventSetTime(type, data, device) {
    if (data.type === 'commandMcuSyncTime' && data.cluster === 'manuSpecificTuya') {
```
or which uses `lib/tuya.js/onEventSetLocalTime`:
```
async function onEventSetLocalTime(type, data, device) {
    // FIXME: What actually nextLocalTimeUpdate/forceTimeUpdate do?
    //  I did not find any timers or something else where it was used.
    //  Actually, there are two ways to set time on TuYa MCU devices:
    //  1. Respond to the `commandMcuSyncTime` event
    //  2. Just send `mcuSyncTime` anytime (by 1-hour timer or something else)
    const nextLocalTimeUpdate = globalStore.getValue(device, 'nextLocalTimeUpdate');
    const forceTimeUpdate = nextLocalTimeUpdate == null || nextLocalTimeUpdate < new Date().getTime();
    if ((data.type === 'commandMcuSyncTime' && data.cluster === 'manuSpecificTuya') || forceTimeUpdate) {
```

Koenkk, 2022-04-25:
The `onEvent` will not conflict with `fz.ignore_tuya_set_time`

Koenkk, 2022-04-28:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4173