Koenkk/zigbee2mqtt #12229: BTicino L4531C (Legrand 067775/741811) only publishes ON/OFF state. No power and voltage

rainbowbreeze, 2022-04-20:
### What happened?

I successfully paired two [BTicino L4531C](https://www.zigbee2mqtt.io/devices/L4531C.html) to z2m in Home Assistant OS environment. The interview was successfull, they were recongnized as [Legrand 067775/741811](https://www.zigbee2mqtt.io/devices/067775_741811.html#legrand-067775%252F741811), but I can only control the ON/OFF state of the two plugs. I have no power or voltage in the state.

```
{
    "linkquality": 255,
    "power": 0,
    "state": "ON",
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "action": null,
    "current": null,
    "voltage": null
}
```

I also deleted and re-paired a couple of times the plugs, but the result was the same.

### What did you expect to happen?

Once paired, I can see state, power and voltage measurements.

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"ixcos-5","value":true}'
Info Zigbee: allowing new devices to join.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"ixcos-5"}'
Info Device '0x0004740000937ec5' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000937ec5","ieee_address":"0x0004740000937ec5"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0004740000937ec5"},"type":"device_connected"}'
Info MQTT publish: topic 'homeassistant/switch/0x0004740000937ec5/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x0004740000937ec5/set","device":{"identifiers":["zigbee2mqtt_0x0004740000937ec5"],"manufacturer":"Legrand","model":"Power socket with power consumption monitoring (067775/741811)","name":"0x0004740000937ec5","sw_version":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!"},"json_attributes_topic":"zigbee2mqtt/0x0004740000937ec5","name":"0x0004740000937ec5","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x0004740000937ec5","unique_id":"0x0004740000937ec5_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0004740000937ec5/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000937ec5"],"manufacturer":"Legrand","model":"Power socket with power consumption monitoring (067775/741811)","name":"0x0004740000937ec5","sw_version":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x0004740000937ec5","name":"0x0004740000937ec5 action","state_topic":"zigbee2mqtt/0x0004740000937ec5","unique_id":"0x0004740000937ec5_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0004740000937ec5/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000937ec5"],"manufacturer":"Legrand","model":"Power socket with power consumption monitoring (067775/741811)","name":"0x0004740000937ec5","sw_version":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x0004740000937ec5","name":"0x0004740000937ec5 power","state_class":"measurement","state_topic":"zigbee2mqtt/0x0004740000937ec5","unique_id":"0x0004740000937ec5_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0004740000937ec5/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000937ec5"],"manufacturer":"Legrand","model":"Power socket with power consumption monitoring (067775/741811)","name":"0x0004740000937ec5","sw_version":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x0004740000937ec5","name":"0x0004740000937ec5 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0x0004740000937ec5","unique_id":"0x0004740000937ec5_voltage_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0004740000937ec5/current/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000937ec5"],"manufacturer":"Legrand","model":"Power socket with power consumption monitoring (067775/741811)","name":"0x0004740000937ec5","sw_version":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x0004740000937ec5","name":"0x0004740000937ec5 current","state_class":"measurement","state_topic":"zigbee2mqtt/0x0004740000937ec5","unique_id":"0x0004740000937ec5_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0004740000937ec5/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000937ec5"],"manufacturer":"Legrand","model":"Power socket with power consumption monitoring (067775/741811)","name":"0x0004740000937ec5","sw_version":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x0004740000937ec5","name":"0x0004740000937ec5 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x0004740000937ec5","unique_id":"0x0004740000937ec5_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0004740000937ec5/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000937ec5"],"manufacturer":"Legrand","model":"Power socket with power consumption monitoring (067775/741811)","name":"0x0004740000937ec5","sw_version":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x0004740000937ec5","name":"0x0004740000937ec5 update state","state_topic":"zigbee2mqtt/0x0004740000937ec5","unique_id":"0x0004740000937ec5_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0004740000937ec5/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000937ec5"],"manufacturer":"Legrand","model":"Power socket with power consumption monitoring (067775/741811)","name":"0x0004740000937ec5 [... cut ...]
Debug Received Zigbee message from '0x0004740000939ee8', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":54598400,"imageType":17,"manufacturerCode":4129}' from endpoint 1 with groupID null
Debug Device '0x0004740000939ee8' requested OTA
Debug Received Zigbee message from '0x0004740000939ee8', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":54598400,"imageType":17,"manufacturerCode":4129}' from endpoint 1 with groupID null
Debug Device '0x0004740000939ee8' requested OTA
```

rainbowbreeze, 2022-05-23:
Is there any debug data I can provide to investigating the issue?

Koenkk, 2022-05-23:
Can you try to reconfigure the device by pressing on the yellow refresh icon in the frontend?

rainbowbreeze, 2022-05-23:
Thanks for the reply. I did 3-4 times at least (complitely remove the device, and re-configure it). Plus, I've two plugs in the same situation.
Happy to provide other logs I collected over time, but the end result was the same, and also the log messages, from what I saw.

One thing I notice is that the FW version of the plug seems newer than the one on z2m -> this could be something to look at?

rainbowbreeze, 2022-05-23:
For the 2 plugs I have:
```
{"id":16,"type":"Router","ieeeAddr":"0x0004740000937ec5","nwkAddr":19733,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,6,5,64513,2820],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"zclVersion":2}},"genOnOff":{"attributes":{"onOff":1}},"haElectricalMeasurement":{"attributes":{"activePower":0,"acVoltageMultiplier":0,"acVoltageDivisor":0,"acCurrentMultiplier":0,"acCurrentDivisor":0,"acPowerMultiplier":1,"acPowerDivisor":1}},"genIdentify":{"attributes":{"identifyTime":60}}},"binds":[{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00212effff05fa4c","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05fa4c","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff05fa4c","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}}},"appVersion":0,"stackVersion":65,"hwVersion":1,"dateCode":" \u000020210412\u0000\u0000\u0000\u0000\u0000","swBuildId":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!","zclVersion":2,"interviewCompleted":true,"meta":{"configured":-1451534777},"lastSeen":1653320164272,"defaultSendRequestWhen":"immediate"}
{"id":17,"type":"Router","ieeeAddr":"0x0004740000939ee8","nwkAddr":8128,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,6,5,64513,2820],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"zclVersion":2}},"genOnOff":{"attributes":{"onOff":1}},"haElectricalMeasurement":{"attributes":{"activePower":0,"acVoltageMultiplier":0,"acVoltageDivisor":0,"acCurrentMultiplier":0,"acCurrentDivisor":0,"acPowerMultiplier":1,"acPowerDivisor":1}}},"binds":[{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00212effff05fa4c","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05fa4c","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff05fa4c","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}}},"appVersion":0,"stackVersion":65,"hwVersion":1,"dateCode":" \u000020210521\u0000\u0000\u0000\u0000\u0000","swBuildId":"01b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!","zclVersion":2,"interviewCompleted":true,"meta":{"configured":-1451534777},"lastSeen":1653319913989,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2022-05-23:
Can you try to change the `repChange` of the `activePower` attribute to `0`, max to 5 and min to 1 in the reporting tab of the frontend?

rainbowbreeze, 2022-05-23:
I'd love to change it in the frontend. But https://github.com/Koenkk/zigbee2mqtt/issues/12575 prevents me from doing it. Happy to do it in another way, if any.

rainbowbreeze, 2022-05-23:
Thanks to a workaround I've find for the afore mentioned bug, I was able to open the frontend, and I changed the values you suggested.

So, I change from this
[image]

To this
[image]

Unfortunately, still no power measurement.

Looking at the log, now every 4-5 secs I get this message
```
debug 2022-05-24 01:49:41Received Zigbee message from '0x0004740000939ee8', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
info 2022-05-24 01:49:41MQTT publish: topic 'zigbee2mqtt/0x0004740000939ee8', payload '{"action":null,"current":null,"linkquality":240,"power":0,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
```
While, previosly, the frequency was way less often.

Any suggestion?

Koenkk, 2022-05-24:
Looking at your logs it seems that the device is malfunctioning (I assume you connected a load to the plug). You can see the device reports 0 power: `'{"activePower":0}'`.

Koenkk, 2022-05-24:
> Could it be they're both malfunctioning? 

I don't see another reason why they wouldn't work, the only thing to make this sure is to test if these devices works with the original hub.

> I also see there is no "current" value too... or historical power consumption of any sort. Is this expected?

Yes, these values are provided by the `seMetering` cluster which this device does not support

rainbowbreeze, 2022-05-27:
I reset and repaired once more both of the plugs... And this time I'm able to read the power!

BTW, for the power consumption measurement, knowing it's not supported by the device, I'll use the [Riemann sum integral Home Assistant integration](https://www.home-assistant.io/integrations/integration/#energy) to create an energy sensor. Slightly off topic, but it could be useful to future thread readers.

Thanks @Koenkk for all the support provided.

closed 2022-05-27 by rainbowbreeze