Koenkk/zigbee2mqtt #12230: Yale YRD226 YRD226HA2619 is not reporting action_source_user correctly

zeiphon, 2022-04-20:
### What happened?

When an action is performed on the lock, `action_source_user` of the entity state is not updated to reflect the user ID performing the action. It remains null.
However, `action_user` *is* updated to the correct user ID, but this is not included as part of the exposed home assistant entity:

``` 
{ "battery": 100, 
"linkquality": 255, 
"lock_state": "unlocked", 
"state": "UNLOCK", 
"action": null, 
"action_source": 0, 
"action_source_name": null, 
"action_source_user": null, 
"action_user": 2 }
```

### What did you expect to happen?

`action_source_user` should be set to the user ID performing the action on the lock (e.g. keypad unlock).

### How to reproduce it (minimal and precise)

Performing any action with the keypad on the lock.

### Zigbee2MQTT version

1.25.0-1 

### Adapter firmware version

0x26660700

### Adapter

ConBee2

wixoff, 2022-04-28:
I am seeing this exact same issue on my Kwikset/Weiser SmartCode 10 (9GED18000-009) lock, confirmed by watching the raw MQTT messages.

I'm using the Sonoff Zigbee 3.0 USB dongle with firmware 20211217 and Z2M version 1.25.0.

zeiphon, 2022-05-29:
Still active

zeiphon, 2022-06-29:
Still active

ouchkilljoys, 2022-07-08:
Yeah I'm seeing the same issue with my lock YRD226 how would we go about changing this?

zeiphon, 2022-08-09:
> Yeah I'm seeing the same issue with my lock YRD226 TSDB how would we go about changing this?

I would like to know as well, I'm not familiar enough with the underlying workings to fix this myself

ouchkilljoys, 2022-08-11:
> > Yeah I'm seeing the same issue with my lock YRD226 TSDB how would we go about changing this?
> 
> I would like to know as well, I'm not familiar enough with the underlying workings to fix this myself

If youve updated to the latest it should be fixed, there is another attribute which should correctly report the user that did the unlock action

closed 2022-09-18 by github-actions[bot]

mghersi65, 2022-10-04:
Hi, Yale226 can be open and close from Home Assistant or MQTT using Zigbee ? Thanks