Koenkk/zigbee2mqtt #12234: ROB_200-017-1 - UNSUPPORTED_ATTRIBUTE

Chamondgl, 2022-04-20:
### What happened?

Pairing a new device, identical in version number and firmware version fails.

### What did you expect to happen?

A successful pairing, just like the two devices prior.

### How to reproduce it (minimal and precise)

Pairing/reconfiguring. The two other devices provide errors too while working OK previously.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20200417

### Adapter

usb-Texas_Instruments_XDS110__03.00.00.12__Embed_with_CMSIS-DAP_L1100F9Q-if00

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8525099/log.txt)

Chamondgl, 2022-04-21:
Confirmed and tested on dev branch. Sorry for wasting your time. I'll be sure to NOT skip the dev branch step in the issue reporting steps next time ;-)

Out of curiosity and to possibly trouble shoot this in the future, what was the issue and how was it fixed?

Chamondgl, 2022-04-21:
Fixed in Dev branch.

closed 2022-04-21 by Chamondgl

Koenkk, 2022-04-22:
This was fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4086