Koenkk/zigbee2mqtt #12235: [Feature request]: Option to enable sensor.<device>_last_seen entity within HA from Z2M configuration settings

jarrah31, 2022-04-20:
### Is your feature request related to a problem? Please describe

Based on the latest update to the following HA Blueprint, I need to enable the last_seen entity on my devices so that I can be notified when they go offline. 
https://community.home-assistant.io/t/offline-detection-for-z2m-devices-with-last-seen/310138
The problem is I have 125 devices, so manually enabling last_seen for each one will take far too long!
[image]

### Describe the solution you'd like

Related to this comment https://github.com/Koenkk/zigbee2mqtt/issues/8287#issuecomment-939254755, would it be possible to have an option that enables last_seen entities by default please? Perhaps called something like `last_seen_entity'
```
advanced:
  # Optional: Add a last_seen attribute to MQTT messages, contains date/time of last Zigbee message
  # possible values are: disable (default), ISO_8601, ISO_8601_local, epoch (default: disable)
  last_seen: 'ISO_8601_local'
  last_seen_entity: 'true'
```

### Describe alternatives you've considered

Other than manually enabling last_seen for all 125 devices, I'm not aware of an alternative method.

### Additional context

Another user described the same problem in a similar closed thread:
https://github.com/Koenkk/zigbee2mqtt/issues/9398#issuecomment-1003752055

jarrah31, 2022-04-20:
Thanks to the Blueprint author, he suggested the following, but unfortunately this didn't work for me, even after restarting Z2M and HA.
```
device_options:
  homeassistant:
    last_seen:
      enabled_by_default: true
```
Just in case anything is clashing, here is my config:
```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbeepi2
  server: mqtt://192.168.0.10
  user: user
  password: pass
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: /dev/ttyUSB0
frontend:
  port: 8080
device_options:
  legacy: false
  homeassistant:
    last_seen:
      enabled_by_default: true
experimental:
  new_api: true
advanced:
  network_key:
    - removed
  pan_id: removed
  channel: 23
  last_seen: ISO_8601_local
  elapsed: true
devices:
```

Koenkk, 2022-04-21:
> I need to enable the last_seen entity on my devices so that I can be notified when they go offline.

You should use the availability feature for this instead: https://www.zigbee2mqtt.io/guide/configuration/device-availability.html

jarrah31, 2022-04-21:
Hey @Koenkk, thank you for the suggestion. Would we then monitor the device's online/offline status within Home Assistant?

Koenkk, 2022-04-22:
Yes, since offline devices will be marked as unavailable in HA.

jarrah31, 2022-04-22:
Thanks Koenkk!

jarrah31, 2022-04-22:
Sorry, one more question @Koenkk, what is the best way to identify all devices from Z2M within Home Assistant please?

I see that historically, the following has been used, but my devices don't appear to have this attribute anymore.
```
{% for state in states.sensor | selectattr('attributes.unit_of_measurement', '==', 'lqi') %}
```
If I could enable all my _last_seen entities, the following may be a possibility, but this also retrieves entities unrelated to Z2M such as my Alexa devices
```
{% for state in states.sensor | selectattr('attributes.device_class', '==', 'timestamp') %}
```
So is there a reliable way to search all Z2M devices so that I can report on any that are unavailable please?

Koenkk, 2022-04-22:
You can use something like https://community.home-assistant.io/t/unavailable-entity-detection-notification/337272

jarrah31, 2022-04-22:
Thank you for the suggestion, but unfortunately this picks up all my unavailable entities (around 40) whereas I only have one unavailable Z2M entity.

If possible I'd like to figure out why the following doesn't enable all my last_seen entities in HA please because this would give me a useful set of Z2M entities to monitor.
```
device_options:
  homeassistant:
    last_seen:
      enabled_by_default: true
```
Within MQTT Explorer I can see that the sensor reported by HA has `enabled_by_default` set to true:
[image]

However, the entity remains disabled within HA:
[image]

I'll be happy to run any further tests to help troubleshoot this issue.

Koenkk, 2022-04-22:
> If possible I'd like to figure out why the following doesn't enable all my last_seen entities in HA please because this would give me a useful set of Z2M entities to monitor.

Because home assistant doesn't update this after initial discovery. So it should work if you start z2m first with `homeassistant: false` and then enable it again, however this will reset all your entities in HA (icons, friendly names)

jarrah31, 2022-04-22:
aha, thank you!

Umm, what is the correct restart order to make this work please?

I've tried the following so far with no luck:
- Set `homeassistant: false`, restarted Z2M, set `homeassistant: true`, restarted Z2M
- Set `homeassistant: false`, restarted Z2M, restarted HA (devices still visible), homeassistant: true, restarted Z2M, last_seen still hidden.

- In the meantime I will try shutting down HA, set `homeassistant: false`, restart Z2M, start HA, set `homeassistant: true`, restart Z2M

jarrah31, 2022-04-22:
The last step I tried didn't work either, however the following has now worked!

1. Stop Z2M, set `homeassistant: false`
2. remove MQTT integration in HA
3. restart HA, confirmed no devices
4. start Z2M
5. stop Z2m
6. set `homeassistant: true`
7. start Z2m
8. add and reconfigure MQTT integration in HA. 

I suspect I could just do the following though and not set homeassistant to false in Z2m?
1. stop Z2M
2. remove MQTT integration in HA
3. restart HA
4. start Z2M
5. add and reconfigure MQTT integration in HA

Koenkk, 2022-04-23:
> I suspect I could just do the following though and not set homeassistant to false in Z2m?

Yes that should also work. Great that you've found a solution.

closed 2022-04-23 by Koenkk

iz3man, 2022-09-09:
> I suspect I could just do the following though and not set homeassistant to false in Z2m?

stop Z2M
remove MQTT integration in HA
restart HA
start Z2M
add and reconfigure MQTT integration in HA
restart HA

This does/can not work, as Z2M will not start w/o MQTT availabilty. Just tried myself. :(

popy2k14, 2023-04-03:
Had the same issue. An easy way is:

- Go to Entities list in HA
- select "Show deactivated entities" on the 3 dot menu
- search for "last_seen"
- select all
- press "Activate entities"

Dont know if this option was available when you guys had the issue.

eitel13, 2023-06-18:
Maaaan, so glad I found this (especially the suggestion from @popy2k14 🥳 ) buuuut (coming from a slightly ignorant POV) why aren't the entities for `last_seen` + `linkquality` automagically enabled in HA? 🤔

popy2k14, 2023-06-18:
@eitel13 youre Welcome
That was the same question I was asking myself 🙂

florian-asche, 2023-08-20:
I have to say, this is so fucked up. Sorry for my words, but to enable last_seen on 60+ devices i have to delete them all and recreate them in homeassistant?

popy2k14, 2023-08-20:
@florian-asche have you seen my post above: https://github.com/Koenkk/zigbee2mqtt/issues/12235#issuecomment-1494734266
You can do this in batch in HA.

florian-asche, 2023-08-20:
> @florian-asche have you seen my post above: [#12235 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/12235#issuecomment-1494734266) You can do this in batch in HA.

Ahh! Sorry, no i did not see that after hours of search for a solution. Will try that thx.

popy2k14, 2023-08-20:
You're welcome :-)

schollex, 2023-11-28:
My way to identify all Z2M-entities which are unavailable:

```
{% set result = states.sensor
      | selectattr('attributes.unit_of_measurement', '==', 'lqi')
      | selectattr('state', '==', 'unavailable')
      | map(attribute='name')
      | list %}
{% set entities = result|join('\n- ') %}
{{ entities }}
```

xelemorf, 2024-04-03:
> My way to identify all Z2M-entities which are unavailable:
> 
> ```
> {% set result = states.sensor
>       | selectattr('attributes.unit_of_measurement', '==', 'lqi')
>       | selectattr('state', '==', 'unavailable')
>       | map(attribute='name')
>       | list %}
> {% set entities = result|join('\n- ') %}
> {{ entities }}
> ```
> 
> Or via device-name:
> 
> ```
> {%set result = namespace(devices=[]) %} 
> 
> {%set entities = states.sensor
>   | selectattr('attributes.unit_of_measurement', '==', 'lqi')
>   | selectattr('state', '==', 'unavailable')
>   | map(attribute='entity_id')
>   | list %}
> {%for entity_id in entities %}
>   {% set result.devices = result.devices + [device_attr(entity_id, 'name')] %}
> {%endfor%}
> 
> {%if result.devices != [] %}
>   {{ '- '+result.devices|join('\n- ') }}
> {%endif%}
> ```

May I ask for some clarification how to incorporate this into an automation?