Koenkk/zigbee2mqtt #12236: [New device support]:

madfx-cpu, 2022-04-21:
### Link

https://www.philips-hue.com/en-au/search?query=7602031PU

### Database entry

{"id":14,"type":"Router","ieeeAddr":"0x001788010b436ab6","nwkAddr":53374,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"7602031PU","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"7602031PU","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20211210","swBuildId":"1.93.7"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20211210","swBuildId":"1.93.7","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1650532163612,"defaultSendRequestWhen":"immediate"}

### Comments

This is an updated model to an already supported device. I tried to add as new but got stuck at section 3.

Koenkk, 2022-04-21:
This device supports color temperature, so please fill it in (https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device)

madfx-cpu, 2022-04-21:
Apologies, I have attempted to read the color temp attributes but am getting the following error:

Warning Received message from unsupported device with Zigbee model '7602031PU' and manufacturer name 'Philips'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Koenkk, 2022-04-22:
Can you make a screenshot of the z2m ui after following: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device ?

madfx-cpu, 2022-04-23:
I've cooked it unfortunately, deleted the pairing to start again figuring I'd messed up the first step and now can't get the damn thing back into pairing mode 🤣 

Think I'll give up on this device, sorry to waste your time!

closed 2022-04-23 by madfx-cpu

Koenkk, 2022-04-23:
See https://www.zigbee2mqtt.io/devices/7146060PH.html how to pair it:

> For the 7146060PH (Philips Hue Go), the power cord has to be connected, after the blinking light (INSTEAD of step four in the video), press and keep holding the button on the bottom until the device is paired (+- 60 seconds). While holding the button the Hue Go will give you a nice light show 😄.