Koenkk/zigbee2mqtt #12241: Heiman HS2SK fails to report power, current and voltage.

KBessmann, 2022-04-21:
### What happened?

I have recently changed from deconz to z2m and i am in the progress to connect all the devices again.
Upon pairing the Heiman HS2SK to z2m, it fails to report the sensors.

### What did you expect to happen?

I expected it to work flawless beeing on the supported devices list.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

0x26720700

### Adapter

Conbee II

### Debug log

Zigbee2MQTT:info  2022-04-21 20:29:28: Configuring 'heiman_switch_fryser'
Zigbee2MQTT:info  2022-04-21 20:29:28: MQTT publish: topic 'zigbee2mqtt/heiman_switch_fryser', payload '{"current":null,"linkquality":255,"power":null,"state":"OFF","voltage":null}'
Zigbee2MQTT:error 2022-04-21 20:29:28: Failed to configure 'heiman_switch_fryser', attempt 3 (Error: Read 0x000d6f0014baea2e/1 haElectricalMeasurement(["acFrequencyDivisor","acFrequencyMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:25:5)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/heiman.js:48:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))

smulle48, 2022-04-22:
Don't think the fix made it into the latest release.

You can use the 1.25.0-dev release. It's fixed there :-)

https://github.com/Koenkk/zigbee2mqtt/issues/12000

Koenkk, 2022-04-22:
Should be fixed in the latest dev indeed (and the next release on 1 May) (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-22 by Koenkk