Koenkk/zigbee2mqtt #12246: Issue with ST218 stelpro heater

onixx, 2022-04-22:
### What happened?

The ST218 heater is supported by stelpro.js. However, there is a problem when this device announces its zigbeeModel string. It seems like there is some extra **random** garbage added to the zigbeeModel when pairing. It varies by device and is not practical as when pairing, sometimes an external converter is needed to support some units.

Here's some info form my database.db
the manufName has a few null chars at the end, I think this is handled already in the code to ignore those.
"manufName":"Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"

However, the zigbeeModel is garbled. Always starts with ST218 but not all chars after are null and they are random. This is an extract from database.db for 6 of those devices
"modelId":"ST218\u0000\u0000\u0000(\u0001\u0000(\u0000\u0000\u0000\u0000\u0000\u0007\u0000�r\u0001\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
"modelId":"ST218\u0000\u0000�r\u0001\u0000�\u0000 \u0001\u0011\u0000 \u0001\u0011\u0015\u0000\u0001\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
"modelId":"ST218\u0000\u0000\u0000(\u0001\u0000(\u0000\u0000\u0000\u0000\u0000\u0007\u0000�r\u0001\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
"modelId":"ST218\u0000\u0000\u0000(\u0001\u0000(\u0000�A\u0000\u0000�_\u0004\u0001\u000e\r\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
"modelId":"ST218\u0000\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0000�r\u0001\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
"modelId":"ST218\u0000\u0000�\u0000 \u0001\u0011\u0000 \u0001\u0011\u0015\u0000\u0001\u0011\u0015 \u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"

When I go to the web interface, the problem is easy to see under the about tab:
By vendor summary:
Stelpro: 6
By model summary:
[image]

In the end, and I cannot explain why, only 2 of those devices needed an external converter to work; these 2
ST218\u0000\u0000\u0000(\u0001\u0000(\u0000�A\u0000\u0000�_\u0004\u0001\u000e\r\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000
ST218\u0000\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0000�r\u0001\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000

### What did you expect to happen?

Is there a way to wildcard in stelpro.js to handle this device's problem with modelid ?

### How to reproduce it (minimal and precise)

Pairing the Stelpro ST218 produces those random modelID

### Zigbee2MQTT version

1.24.0 commit: 7a2ddf2

### Adapter firmware version

20211217

### Adapter

Electrolama

onixx, 2022-04-22:
Issue https://github.com/Koenkk/zigbee2mqtt/issues/810 was an attempt to fix this. Looks like in my case it is not working

onixx, 2022-04-22:
@Koenkk seems like this change is not in zigbee-herdsmans findByDevice

closed 2022-05-30 by github-actions[bot]

bwrsandman, 2023-05-20:
I am also seeing this on 1.30.3. Looks like a regression. Can a dev comment on how to fix this or give a workaround?

[image]

in database.db
```
ST218\u0000\u0000\u0000(\u0001\u0000(\u0000�A\u0000\u0000�_\u0004\u0001\u000e\r\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000
```