Koenkk/zigbee2mqtt #12251: installation ends with warnings and high and critical vulnerabilities

dawiinci, 2022-04-23:
### What happened?

**Installed zigbee2mqtt on macOS**

```
git clone https://github.com/Koenkk/zigbee2mqtt.git ~/zigbee2mqtt
cd ~/zigbee2mqtt
python3 -m venv .

source ~/zigbee2mqtt/bin/activate
pip install --upgrade pip wheel setuptools
pip install nodeenv
nodeenv -p -n lts
deactivate

source~/zigbee2mqtt/bin/activate
cd ~/zigbee2mqtt

npm ci
deactivate
```

Result of npm ci:

```
npm WARN old lockfile 
npm WARN old lockfile The npm-shrinkwrap.json file was created with an old version of npm,
npm WARN old lockfile so supplemental metadata must be fetched from the registry.
npm WARN old lockfile 
npm WARN old lockfile This is a one-time fix-up, please be patient...
npm WARN old lockfile 

added 977 packages, and audited 978 packages in 8s

75 packages are looking for funding
  run `npm fund` for details

2 vulnerabilities (1 high, 1 critical)

To address all issues, run:
  npm audit fix

Run `npm audit` for details.
```

### What did you expect to happen?

Install with no errors

### How to reproduce it (minimal and precise)

Run the command above.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

0x26580700

### Adapter

ConBee II

Koenkk, 2022-04-23:
Seems to be already fixed in the dev branch:

[image]

As dependencies get outdated in stable z2m version, these warnings are inevitable.  Its not something to worry about, just make sure to keep your z2m up-to-date.

dawiinci, 2022-04-23:
Thanks for the quick clarification.

closed 2022-04-23 by dawiinci