Koenkk/zigbee2mqtt #12252: [New device support]: Orvibo DS20Z07B - zigbee model 4a33f5ea766a4c96a962b371ffde9943

RatheeM, 2022-04-23:
### Link

http://orvibo.ae/s/

### Database entry

const definition = {     zigbeeModel: ['4a33f5ea766a4c96a962b371ffde9943'],     model: 'DS20Z07B',     vendor: 'ORVIBO',     description: 'Downlight (S series)',     fromZigbee: [], // We will add this later     toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).     extend: extend.light_onoff_brightness_colortemp({colorTempRange: [166, 370]}), };

### Comments

Paired the device to Z2M - pair successfull but unsupported

Created orvibo.js in zigbee2mqtt folder next to config.yaml with the following:
const definition = {
    zigbeeModel: ['4a33f5ea766a4c96a962b371ffde9943'],
    model: 'DS20Z07B',
    vendor: 'ORVIBO',
    description: 'Downlight (S series)',
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [166, 370]}),
};

Added orvibo.js to external_connector in config.yaml, and changed log level to debug.

Now unsupported tag disappeared from Z2M UI, exposes are visible. But when I execute anything it says"
No converter available - in the log

I could not understand the further instructions in Z2M website regarding adding converter. 

Please help. 

Hardware:
HA with Supervisor on Debian 11 - HA supervised install
Z2M through Conbee II

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4a33f5ea766a4c96a962b371ffde9943'],
    model: 'DS20Z07B',
    vendor: 'ORVIBO',
    description: 'Downlight (S series)',
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [166, 370]}),
};

module.exports = definition;
```

### Supported color modes

onoff_brightness_colortemp

### Color temperature range

166-370 probably. Not sure. The min max through Dev Console - Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":0,"colorTempPhysicalMax":65279}

Koenkk, 2022-04-24:
Try removing:

```js
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
```

RatheeM, 2022-04-24:
Thanks for your reply. Already tried that. Does not work.

Koenkk, 2022-04-24:
I'm quite certain it should work, added it directly now can you check the latest dev branch?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

RatheeM, 2022-04-24:
Controls still do not work.

I have shifted to Z2M Edge.

Log:
Debug 2022-04-24 23:16:37Received MQTT message on 'zigbee2mqtt/0x847127fffe6519ec/set' with data '{"state":"OFF"}'
Debug 2022-04-24 23:16:37Publishing 'set' 'state' to '0x847127fffe6519ec'
Warning 2022-04-24 23:16:41Failed to ping '0x847127fffe6519ec' (attempt 1/1, Read 0x847127fffe6519ec/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received))

RatheeM, 2022-04-24:
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt:
  server: mqtt://core-mosquitto
serial:
  port: /dev/ttyACM0
  adapter: deconz
external_converters:
  - orvibo.js

RatheeM, 2022-04-24:
Orvibo.js is
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4a33f5ea766a4c96a962b371ffde9943'],
    model: 'DS20Z07B',
    vendor: 'ORVIBO',
    description: 'Downlight (S series)',
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [166, 370]}),
};

module.exports = definition;`

RatheeM, 2022-04-24:
Working now. Thanks.

closed 2022-06-01 by github-actions[bot]

hirzel23, 2022-07-14:
hi @RatheeM may I ask? is it all working fine include adjusting the brightness?

RatheeM, 2022-07-14:
Yes. It’s working wonderful.

RatheeM, 2022-10-11:
> thanks for your confirmation, are you using dev or the official z2m?
> are you still using the following js?
> I'm currently use Orivbo DS20Z, I use the same code and seems running
> without dimmer function. the dimmer is still not working. so I'm just
> curious if I can apply the same

Dev.