Koenkk/zigbee2mqtt #12265: IKEA LED2005R5 says not supported when pairing for first time.

vandad51, 2022-04-24:
### What happened?

I purchased a LED2005R5 from IKEA and tried to pair with my HUSBZB-1 zigbee stick. It paired right aways with no hiccups. But after repairing 3-4 times, it would keep saying that it is not supported (see logs). Even though [this page says it is supported](https://www.zigbee2mqtt.io/devices/LED2005R5.html).

### What did you expect to happen?

I expected the device to be supported. But also, I expected the model ID to be LED2005R5 not like it detected upon pairing, "TRADFRIbulbGU10WS380lm" Z2M also didn't recognize the Manufacturer name. Seems like maybe there needs to be some allowances for updated names and IDs that IKEA is now making for some of their devices?

### How to reproduce it (minimal and precise)

After pairing, Z2M says device is not supported.

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

6.7.8.0 build 373

### Adapter

HUSBZB-1 Nortek GoControl

### Debug log

`info  2022-04-23 20:43:22: Zigbee: allowing new devices to join.
info  2022-04-23 20:43:22: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"cak5g-1"}'
info  2022-04-23 20:43:25: Device '0x540f57fffe1db29a' joined
info  2022-04-23 20:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x540f57fffe1db29a","ieee_address":"0x540f57fffe1db29a"},"type":"device_joined"}'
info  2022-04-23 20:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x540f57fffe1db29a"},"type":"device_connected"}'
info  2022-04-23 20:43:25: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:25-05:00"}'
info  2022-04-23 20:43:25: Starting interview of '0x540f57fffe1db29a'
info  2022-04-23 20:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x540f57fffe1db29a","ieee_address":"0x540f57fffe1db29a","status":"started"},"type":"device_interview"}'
info  2022-04-23 20:43:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x540f57fffe1db29a"},"type":"pairing"}'
info  2022-04-23 20:43:26: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:26-05:00"}'
info  2022-04-23 20:43:26: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:26-05:00"}'
info  2022-04-23 20:43:26: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:26-05:00"}'
info  2022-04-23 20:43:27: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:27-05:00","linkquality":255}'
info  2022-04-23 20:43:28: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:28-05:00","linkquality":255}'
info  2022-04-23 20:43:28: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:28-05:00","linkquality":255}'
info  2022-04-23 20:43:28: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:28-05:00","linkquality":255}'
info  2022-04-23 20:43:28: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:28-05:00","linkquality":255}'
info  2022-04-23 20:43:28: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:28-05:00","linkquality":255}'
info  2022-04-23 20:43:28: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:28-05:00","linkquality":255}'
info  2022-04-23 20:43:29: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:29-05:00","linkquality":255}'
info  2022-04-23 20:43:29: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:29-05:00","linkquality":255}'
info  2022-04-23 20:43:29: Successfully interviewed '0x540f57fffe1db29a', device has successfully been paired
warn  2022-04-23 20:43:29: Device '0x540f57fffe1db29a' with Zigbee model 'TRADFRIbulbGU10WS380lm' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-04-23 20:43:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x540f57fffe1db29a","ieee_address":"0x540f57fffe1db29a","status":"successful","supported":false},"type":"device_interview"}'
info  2022-04-23 20:43:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x540f57fffe1db29a","supported":false},"type":"pairing"}'
warn  2022-04-23 20:43:55: Received message from unsupported device with Zigbee model 'TRADFRIbulbGU10WS380lm' and manufacturer name 'IKEA of Sweden'
warn  2022-04-23 20:43:55: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-04-23 20:43:55: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:55-05:00","linkquality":255}'
warn  2022-04-23 20:43:58: Received message from unsupported device with Zigbee model 'TRADFRIbulbGU10WS380lm' and manufacturer name 'IKEA of Sweden'
warn  2022-04-23 20:43:58: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-04-23 20:43:58: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:43:58-05:00","linkquality":241}'
warn  2022-04-23 20:44:01: Received message from unsupported device with Zigbee model 'TRADFRIbulbGU10WS380lm' and manufacturer name 'IKEA of Sweden'
warn  2022-04-23 20:44:01: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-04-23 20:44:01: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:44:01-05:00","linkquality":255}'
warn  2022-04-23 20:44:04: Received message from unsupported device with Zigbee model 'TRADFRIbulbGU10WS380lm' and manufacturer name 'IKEA of Sweden'
warn  2022-04-23 20:44:04: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-04-23 20:44:04: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:44:04-05:00","linkquality":255}'
warn  2022-04-23 20:44:07: Received message from unsupported device with Zigbee model 'TRADFRIbulbGU10WS380lm' and manufacturer name 'IKEA of Sweden'
warn  2022-04-23 20:44:07: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-04-23 20:44:07: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe1db29a', payload '{"last_seen":"2022-04-23T20:44:07-05:00","linkquality":255}'`

vandad51, 2022-04-24:
I am also using Z2M with Home Assistant on VirtualBox

Koenkk, 2022-04-24:
`TRADFRIbulbGU10WS380lm` was not supported yet, added it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-24 by Koenkk

vandad51, 2022-04-24:
@Koenkk Awesome, thanks! Just a question about the dev branch on HA. Do I need to uninstall or disable the current Z2M and install edge? I would just copy the database and config over to the edge files right? Or how does it work?

Koenkk, 2022-04-25:
Just uninstall and install the edge version, everything is migrated automatically.

vandad51, 2022-04-29:
@Koenkk I found the light does work with Z2M Edge, but barely. It's very slow to respond to light setting changes and gives timeout errors like:

Error 2022-04-29 18:38:24Publish 'set' 'color_temp' to '0x540f57fffe1db29a' failed: 'Error: Command 0x540f57fffe1db29a/1 lightingColorCtrl.moveToColorTemp({"colortemp":250,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 59704 - 1 - 27 - 768 - 11 after 10000ms)'

Koenkk, 2022-04-30:
This looks like some interference issue; https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html#usb-based-adapter

vandad51, 2022-04-30:
@Koenkk  Does that matter if my LQI is around 230-240? I can try getting an extension cable. I currently don't have an issue with my Aqara sensors though.

Koenkk, 2022-05-01:
Now sure if that is reliable, I've never seen such high LQI even when you almost put the device into the coordinator