Koenkk/zigbee2mqtt #12268: Philips Hue Outdoor motion sensor stooped working

artohi, 2022-04-24:
### What happened?

I have two Philips Hue Outdoor motion sensors (9290019758). Both of them have worked just fine. Today I noticed that other sensor did not work anymore. All exposed values are Null. Only link quality value is working. 

I tried to re-pair it but same thing... Only link quality works. I deleted the the device (force delete) but same thing.

After that I plugges my Philips Hue hub to network (I do not use it anymore because of Home Assistant and zigbee2mqtt). Then I paired the motion sensor to Hue hub and it worked fine (zigbee channel is different in Hue hub and zigbee2mqtt). 

Then I deleted the motion sensor from hue and paired it again with zigbee2mqtt but the link quality is only value that is working

### What did you expect to happen?

Motion sensor should work like it worked so far. (Other similar sensor is working fine)

### How to reproduce it (minimal and precise)

I have no idea what has happened

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220103

### Adapter

SONOFF ZigBEE 3.0 USB Dongle Plus

artohi, 2022-04-24:
I tried to change the sensors motion sensitivity and led indication value and I got following error message
[z2m_log_change_motion_sensitivity.txt](https://github.com/Koenkk/zigbee2mqtt/files/8550554/z2m_log_change_motion_sensitivity.txt)
s

artohi, 2022-04-24:
And here is the log when there was first "error" at the log (device left the network)
[z2m_log_device_left_the_network.txt](https://github.com/Koenkk/zigbee2mqtt/files/8550579/z2m_log_device_left_the_network.txt)

lkxnon, 2022-04-25:
Funny that over last 2 days same happened to me. Looks like related to #2693 
I am running zzh! with latest Z2M firmware/software. I put this sensor indoor 2 weeks back just to see if issue with latest FW/software resolve network connectivity issue with this sensor but looks like it happened again :( 

So I still wait some time before hanging it outside on wall....

artohi, 2022-04-25:
> Funny that over last 2 days same happened to me. Looks like related to #2693 I am running zzh! with latest Z2M firmware/software. I put this sensor indoor 2 weeks back just to see if issue with [...]

After I noticed your reply I restarted the zigbee network. I thought that it did not help so I picked up screwdriver and was going to remove the batteries from hue motion sensor.  After first screw the sensor announced itself and not is seems to be working... weird.

`info  2022-04-25 11:18:25: Configuring 'HueMotionOutdoorEtuovi'
debug 2022-04-25 11:18:26: Device 'HueMotionOutdoorEtuovi' announced itself`

closed 2022-06-03 by github-actions[bot]