Koenkk/zigbee2mqtt #12271: Xiaomi (Aqara door sensor E1 MCCGQ14LM) leaving network

ksga, 2022-04-25:
### What happened?

Device pairs, interviews and is recognized, but leave the network immediately

### What did you expect to happen?

Stable connection

### How to reproduce it (minimal and precise)

Remove device from network if present
Permit join
Insert battery into device
Hold switch for 5 seconds

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8552123/log.txt)
The device is 0x54ef441000294a4f

ksga, 2022-04-25:
I decided to reset my network completely. Flashed the coordinator with default IEEE (previously used the IEEE from my old dongle), deleted database.db, coordinator_backup.json, state.json, devices.yaml and groups.yaml

Lights (various Ikea bulbs/driver and Nordtronic dimmer) and a sensor (Xiaomi WSDCGQ11LM) pairs fine - but in addition to the problem with the door sensor all of my Ikea switches (E1743/E1810) won't pair. Debug log from a E1743:
```
info  2022-04-25 11:12:40: Device '0x8cf681fffe55a614' joined
info  2022-04-25 11:12:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe55a614","ieee_address":"0x8cf681fffe55a614"},"type":"device_joined"}'
info  2022-04-25 11:12:40: Starting interview of '0x8cf681fffe55a614'
info  2022-04-25 11:12:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe55a614","ieee_address":"0x8cf681fffe55a614","status":"started"},"type":"device_interview"}'
debug 2022-04-25 11:12:42: Device '0x8cf681fffe55a614' announced itself
info  2022-04-25 11:12:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe55a614","ieee_address":"0x8cf681fffe55a614"},"type":"device_announce"}'
warn  2022-04-25 11:12:50: Device '0x8cf681fffe55a614' left the network
info  2022-04-25 11:12:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe55a614","ieee_address":"0x8cf681fffe55a614"},"type":"device_leave"}'
debug 2022-04-25 11:13:11: Saving state to file /config/zigbee2mqtt/state.json
info  2022-04-25 11:13:22: Device '0x8cf681fffe55a614' joined
info  2022-04-25 11:13:22: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe55a614","ieee_address":"0x8cf681fffe55a614"},"type":"device_joined"}'
info  2022-04-25 11:13:22: Starting interview of '0x8cf681fffe55a614'
info  2022-04-25 11:13:22: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe55a614","ieee_address":"0x8cf681fffe55a614","status":"started"},"type":"device_interview"}'
debug 2022-04-25 11:13:24: Device '0x8cf681fffe55a614' announced itself
info  2022-04-25 11:13:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe55a614","ieee_address":"0x8cf681fffe55a614"},"type":"device_announce"}'
debug 2022-04-25 11:13:27: Received Zigbee message from '0x8cf681fffe55a614', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2022-04-25 11:13:27: Skipping message, definition is undefined and still interviewing
debug 2022-04-25 11:13:29: Received Zigbee message from '0x8cf681fffe55a614', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2022-04-25 11:13:29: Skipping message, definition is undefined and still interviewing
warn  2022-04-25 11:13:33: Device '0x8cf681fffe55a614' left the network
info  2022-04-25 11:13:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe55a614","ieee_address":"0x8cf681fffe55a614"},"type":"device_leave"}'
```
I pressed the up/down button during pairing, and as can be seen from the log above, the command is sent, just not interpreted...

It seems that the interview part is the problem.

ksga, 2022-04-25:
Update

Managed to pair the Ikea switches. Moved them away from the coordinator and close to a paired bulb. They now paired without problems, and are now shown as childs of other devices than the coordinator on the network map.
The door switch is still acting a bit strange...

But maybe it's a coordinator device/firmware bug.

dkouskousis, 2022-04-30:
I have the same problem. :(

ricargr, 2022-05-14:
Same here with Slaesh's CC2652RB stick

ksga, 2022-05-24:
Received a new device from seller - and it works like it should. So maybe different firmware on the sensor, or simply a defective device.
Pairing directly with the Sonoff dongle still producing the join/leave error, but placing it close to an IKEA bulb and pressing the switch a couple of times during pairing did the trick.

ikaruswill, 2022-06-17:
+1 I recently 'upgraded' from a `CC2531` to a `SONOFF Zigbee 3.0 USB Dongle Plus` and my `IKEA E1743` just wouldn't pair. Faced the same issue of it starting the interview and immediately failing after 10s, followed by leaving the network.

I've since switched back to my `CC2531` and the `IKEA E1743` dimmer switches now pair and work just fine.

closed 2022-07-26 by github-actions[bot]

Erelen-Laiquendi, 2022-08-31:
Same problem with Slaesh's CC2652RB stick & Aqara E1 door & window contact sensor (MCCGQ14LM)
:(

Kaptensanders, 2023-02-09:
Same problem for all my Aqara Motion P1's -> Sonoff Dongle-E Router firmware, Z2M

-->device_joined
-->Configuring
--> Successfully configured
10s pass
-->device_leave

I'm playing with the idea that messages are processed or reaches the the coordinator or Z2M in the wrong order. 
For other Aqara devices, when re-pairing (eg long-press reset), there is an initial device_leave instruction followed by device_joined, and the interview... 

This started happening after adding the Sonoff Dongle-E Routers. Before, when pairing directly with the coordinator (also Sonoff Dongle-E), there was no issues.

leeaash, 2024-02-19:
I had the same issue with Aqara Motion P1, any solution/way to resolve that?