Koenkk/zigbee2mqtt #12272: CMD crash with no errors

fernando155, 2022-04-25:
### What happened?

Good afternoon.

I´m using windows and the cmd commands to have zigbee connected, and was working "well" since 4 days ago, that started to close with no reason at all. I know that this way is rubbish, and a long etc, but is how I do it, and until I'm not changing the computer, I don´t want to change it.

The only changes I got from that day was:

Upgrade HA to 2022.4.6 and add two devices: TS0012_switch_module and TS0002. The first onw gives me this:

Zigbee2MQTT:error 2022-04-25 13:18:49: Publish 'set' 'state' to 'F.2 Bath Switch' failed: 'Error: Command 0xa4c138b10a210eb1/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:info  2022-04-25 13:18:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'F.2 Bath Switch' failed: 'Error: Command 0xa4c138b10a210eb1/2 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"F.2 Bath Switch"},"type":"zigbee_publish_error"}'

But I don't know what I can do, I'm just completly lost

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2

fernando155, 2022-04-25:
well, using powershell it give me a fault:

Error: Cluster with name '57344' does not exist
    at Object.getCluster (C:\Zigbee2Mqtt\node_modules\zigbee-herdsman\src\zcl\utils.ts:66:19)
    at Endpoint.saveClusterAttributeKeyValue (C:\Zigbee2Mqtt\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:242:35)
    at Controller.onZclOrRawData (C:\Zigbee2Mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:644:26)
    at DeconzAdapter.<anonymous> (C:\Zigbee2Mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:135:70)
    at DeconzAdapter.emit (node:events:390:28)
    at DeconzAdapter.checkReceivedDataPayload (C:\Zigbee2Mqtt\node_modules\zigbee-herdsman\src\adapter\deconz\adapter\deconzAdapter.ts:1134:22)
    at EventEmitter.<anonymous> (C:\Zigbee2Mqtt\node_modules\zigbee-herdsman\src\adapter\deconz\adapter\deconzAdapter.ts:71:78)
    at EventEmitter.emit (node:events:390:28)
    at parseReadReceivedDataResponse (C:\Zigbee2Mqtt\node_modules\zigbee-herdsman\src\adapter\deconz\driver\frameParser.ts:276:23)
    at parseFrame (C:\Zigbee2Mqtt\node_modules\zigbee-herdsman\src\adapter\deconz\driver\frameParser.ts:458:32)

Koenkk, 2022-04-25:
`Error: Cluster with name '57344' does not exist` -> I think this is already fixed in z2m 1.25.0

fernando155, 2022-04-25:
> `Error: Cluster with name '57344' does not exist` -> I think this is already fixed in z2m 1.25.0

I updated to 1.25.0 and was doing the same. Didn't see if was the same thing. But... I removed the bath switch and is been a few hours stable. Before, after a few hours was quite bad and slow, but now everything looks fine. 

So I think is the switch the problem. I will keep looking. If it goes tomorrow as normal I know that was the device. 

But the device, looks like is only the switch that goes to the light, not the one that goes to the fan. I'm suspicious about that doesn't have neutral...

closed 2022-06-03 by github-actions[bot]