Koenkk/zigbee2mqtt #12279: Xiaomi Aqara S2 lock ZNMS12LM, state ON/OFF instead of LOCK/UNLOCK

rogodra, 2022-04-25:
### What happened?

Since the last update, the lock of Xiaomi Aqara S2 lock, ZNMS12LM, the state after a while changes only to OFF or ON, instead of UNLOCK or LOCK.

That is, if you use the lock, the values are correct, UNLOCK or LOCK, but for some reason, after a while the value changes to OFF or ON.

Causing that when restarting Home Assistant it comes out with Unknown state, because from what I understand, it does not understand the value of OFF or ON.

When the value of "state" is LOCK / UNLOCK everything works fine:
[image]

After a few minutes the state value changes to ON / OFF, so Home Assistant detects it as a "Unknown" value:
[image]

### How to reproduce it (minimal and precise)

The status value only changes from UNLOCK / LOCK to OFF / ON after a while from using the lock. The value of the reverse marks correctly. Problems have been detected since the latest version 1.25, so far it has always been marked correctly.

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

20211116

### Adapter

CC2530

### Debug log

**_Reading before using the Aqara S2:_**
Zigbee2MQTT:info  2022-04-25 19:40:52: MQTT publish: topic 'zigbee2mqtt/Aqara S2 lock', payload '{"action":null,"battery":33,"battery_low":true,"linkquality":null,"reverse":"UNLOCK","state":"OFF","temperature":21,"update":{"state":"idle"},"update_available":false,"voltage":3390}'

**_After using the Aqara S2, mark the correct status as UNLOCK:_**
Zigbee2MQTT:info  2022-04-25 19:42:54: MQTT publish: topic 'zigbee2mqtt/Aqara S2 lock', payload '{"action":"reverse_lock","battery":33,"battery_low":true,"linkquality":31,"reverse":"LOCK","state":"UNLOCK","temperature":21,"update":{"state":"idle"},"update_available":false,"voltage":3390}'
Zigbee2MQTT:info  2022-04-25 19:42:54: MQTT publish: topic 'zigbee2mqtt/Aqara S2 lock', payload '{"action":"","battery":33,"battery_low":true,"linkquality":31,"reverse":"LOCK","state":"UNLOCK","temperature":21,"update":{"state":"idle"},"update_available":false,"voltage":3390}'
Zigbee2MQTT:info  2022-04-25 19:42:54: MQTT publish: topic 'zigbee2mqtt/Aqara S2 lock/action', payload 'reverse_lock'
Zigbee2MQTT:info  2022-04-25 19:42:55: MQTT publish: topic 'zigbee2mqtt/Aqara S2 lock', payload '{"action":"reverse_lock_cancel","battery":33,"battery_low":true,"linkquality":31,"reverse":"UNLOCK","state":"UNLOCK","temperature":21,"update":{"state":"idle"},"update_available":false,"voltage":3390}'
Zigbee2MQTT:info  2022-04-25 19:42:55: MQTT publish: topic 'zigbee2mqtt/Aqara S2 lock', payload '{"action":"","battery":33,"battery_low":true,"linkquality":31,"reverse":"UNLOCK","state":"UNLOCK","temperature":21,"update":{"state":"idle"},"update_available":false,"voltage":3390}'

**_Without touching the Aqara S2, the next reading of the "state" value turns OFF:_**
Zigbee2MQTT:info  2022-04-25 20:14:00: MQTT publish: topic 'zigbee2mqtt/Aqara S2 lock', payload '{"action":null,"battery":33,"battery_low":true,"linkquality":31,"reverse":"UNLOCK","state":"OFF","temperature":21,"update":{"state":"idle"},"update_available":false,"voltage":3390}'

McGiverGim, 2022-04-26:
I thinkI know where they problem is. Can you take a log in debug mode to confirm?

rogodra, 2022-04-26:
Hello, I am attaching the debug data that I have extracted
[Debug Aqara S2.txt](https://github.com/Koenkk/zigbee2mqtt/files/8566736/Debug.Aqara.S2.txt)
.

McGiverGim, 2022-04-27:
Thanks for the log. I've pushed a PR to fix the issue.

Koenkk, 2022-04-27:
@McGiverGim big thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and the next release on 1 May