Koenkk/zigbee2mqtt #12281: Eglo Fueva mis-identified as lightbulb

rnsc, 2022-04-25:
### What happened?

Similar to #11577 , I have a ceiling mounted Eglo that is recognised as a light bulb and doesn't support all the functions of the device.
It's tagged as : [33951/33948](https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948)

Here's a link to the device: https://www.eglo.com/uk/surface-mounted-light-fueva-z-98843.html

If this should be a new device addition, I'd be happy to provide more information if needed.

Here's the log:
```raw
Zigbee2MQTT:info  2022-04-25 16:39:28: Device '0xa4c138f24760f8bb' joined
Zigbee2MQTT:info  2022-04-25 16:39:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f24760f8bb","ieee_address":"0xa4c138f24760f8bb"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-04-25 16:39:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138f24760f8bb"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-04-25 16:39:28: Starting interview of '0xa4c138f24760f8bb'
Zigbee2MQTT:info  2022-04-25 16:39:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f24760f8bb","ieee_address":"0xa4c138f24760f8bb","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-04-25 16:39:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138f24760f8bb"},"type":"pairing"}'
Zigbee2MQTT:info  2022-04-25 16:39:28: Successfully interviewed '0xa4c138f24760f8bb', device has successfully been paired
Zigbee2MQTT:info  2022-04-25 16:39:28: Device '0xa4c138f24760f8bb' is supported, identified as: AwoX LED white (33951/33948)
Zigbee2MQTT:info  2022-04-25 16:39:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"LED white","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"33951/33948","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0}],"supports_ota":false,"vendor":"AwoX"},"friendly_name":"0xa4c138f24760f8bb","ieee_address":"0xa4c138f24760f8bb","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2022-04-25 16:39:28: Configuring '0xa4c138f24760f8bb'
Zigbee2MQTT:info  2022-04-25 16:39:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"LED white","friendly_name":"0xa4c138f24760f8bb","model":"33951/33948","supported":true,"vendor":"AwoX"},"type":"pairing"}'
Zigbee2MQTT:info  2022-04-25 16:39:28: Successfully configured '0xa4c138f24760f8bb'
Zigbee2MQTT:info  2022-04-25 16:39:29: MQTT publish: topic 'zigbee2mqtt/0xa4c138f24760f8bb', payload '{"brightness":254,"linkquality":255}'
Zigbee2MQTT:info  2022-04-25 16:39:29: MQTT publish: topic 'zigbee2mqtt/0xa4c138f24760f8bb', payload '{"brightness":254,"linkquality":255,"state":"ON"}'
Zigbee2MQTT:info  2022-04-25 16:39:45: MQTT publish: topic 'zigbee2mqtt/0xa4c138f24760f8bb', payload ''
```

### What did you expect to happen?

I was expecting to be able to change the colour temperature on the device, but could only switch it ON/OFF or change the brightness.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

rnsc, 2022-04-25:
The EGLO website says that the lamp doesn't support changing the colour temperature, but that's not true, I can change it with the AWOX application:
[image]

rnsc, 2022-04-26:
@Koenkk here's the extract of the db (not sure why the type is Router):

```json
{
  "id": 5,
  "type": "Router",
  "ieeeAddr": "0xa4c138f24760f8bb",
  "nwkAddr": 11791,
  "manufId": 4417,
  "manufName": "AwoX",
  "powerSource": "Mains (single phase)",
  "modelId": "TLSR82xx",
  "epList": [
    1,
    242,
    3
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 268,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        768,
        4096,
        64599,
        10
      ],
      "outClusterList": [
        6
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "appVersion": 0,
            "hwVersion": 0
          }
        },
        "lightingColorCtrl": {
          "attributes": {
            "colorCapabilities": 16,
            "colorTempPhysicalMin": 153,
            "colorTempPhysicalMax": 370
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "3": {
      "profId": 4751,
      "epId": 3,
      "devId": 268,
      "inClusterList": [
        65360,
        65361
      ],
      "outClusterList": [
        65360,
        65361
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 0,
  "stackVersion": 2,
  "hwVersion": 0,
  "swBuildId": "0122052017",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "configured": 88764544
  },
  "lastSeen": 1650899256240,
  "defaultSendRequestWhen": "immediate"
}
```

Koenkk, 2022-04-26:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

rnsc, 2022-04-26:
@Koenkk confirmed, it's working properly now, I can change the colour! Thank you for fixing this super fast!

rnsc, 2022-04-26:
Added a tip through GitHub sponsor!

closed 2022-04-26 by rnsc