Koenkk/zigbee2mqtt #12282: Aqara sensors stopped reporting voltage

protyposis, 2022-04-26:
### What happened?

I use 5 Aqara WSDCGQ11LM temperature sensors. Since their batteries drained and I replaced them with new ones, 2 sensors are exhibiting odd behavior:

 * No voltage reported (all other values are reported: temperature, humidity, pressure)
 * No voltage entities in HA
 * Z2M exposes a static voltage level that seems to be the same since the batteries were changed (3985 and 3005 mV)
 * Resetting/repairing does not bring back voltage reporting

### What did you expect to happen?

I expect voltages being continued to be reported just like they were before the battery replacement.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

0x26720700

### Adapter

 ConBee2

lordblobo, 2022-04-26:
When I installed my Aqara temp/humidity sensors, it took them about a day to start providing the voltage//battery readings. They did not do so from the beginning, maybe because of some averaging algorithm in the firmware.

Could this also be the case after battery replacement?

protyposis, 2022-04-26:
I though so too and waited, but now 15 days have passed and it still didn't start. The other 3 sensors started reporting voltage within a day though.

protyposis, 2022-04-29:
I noticed today that the signal strength isn't reported either and found the solution. The voltage and signal strength entities were deactivated in HA, thus no values were recorded! Seems like they automatically deactivated when the batteries were empty. Additionally, the voltage never changed during the time I monitored it manually, which led me believe its already failing at the hardware or Z2M level.

closed 2022-04-29 by protyposis