Koenkk/zigbee2mqtt #12285: [New device support]: Tuya Thermometer/hygrometer with LCD _TZE200_locansqn

lowerprofile, 2022-04-26:
### Link

https://www.aliexpress.com/item/1005003634353180.html

### Database entry

{"id":6,"type":"EndDevice","ieeeAddr":"0xa4c13860345c1a64","nwkAddr":45366,"manufId":4417,"manufName":"_TZE200_locansqn","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000e��)i\u0018��)i\"��)i,��)i6��)i@��)iO��)iP��)i","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b7799c","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1650979439722,"defaultSendRequestWhen":"immediate"}

### Comments

I bought a couple of these from Ali expecting them to work as a generic Tuya temp/humidity indicator. I did get temp, humidity and battery but I also got a load of no converter messages, so I got hold of a bridge and did a little digging.

Once I had the datapoints and had added to & from converters it all works nicely with the exception that changing settings (report interval, temperature alarm points etc) seems to have no effect. They are all read back but when I make changes in the frontend nothing seems to happen.

It's my first poke at creating the converters so I wondered if someone could take a look and tell me where I'm going wrong....

Datapoints:
`  //KSMT Smart Zigbee Temperature and Humidity Display datapoints
    ksmtTemperature: 1,
    ksmtHumidity: 2,
    ksmtBattery: 4,
    ksmtTempUnit: 9,
    ksmtMaxTemp: 10,
    ksmtMinTemp: 11,
    ksmtMaxHumidity: 12,
    ksmtMinHumidity: 13,
    ksmtTempAlarm: 14,
    ksmtHumidityAlarm: 15,
    ksmtTempReport: 17,
    ksmtHumidityReport: 18,
    ksmtTempSensitivity: 19,
    ksmtHumiditySensitivity: 20,`

FromZigbee converter:
`ksmt_lcd_temperature_humidity_sensor: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'commandDataResponse'],
        options: [exposes.options.precision('temperature'), exposes.options.calibration('temperature'),
            exposes.options.precision('humidity'), exposes.options.calibration('humidity')],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'ksmt_lcd_temperature_humidity_sensor');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            switch (dp) {
            case tuya.dataPoints.ksmtTemperature:
                return {temperature: calibrateAndPrecisionRoundOptions(value / 10, options, 'temperature')};
            case tuya.dataPoints.ksmtHumidity:
                return {humidity: calibrateAndPrecisionRoundOptions(value, options, 'humidity')};
            case tuya.dataPoints.ksmtBattery:
                return {battery: value};
            case tuya.dataPoints.ksmtTempUnit:
                return {temperature_unit_convert: {0x00: 'celsius', 0x01: 'fahrenheit'}[value]};
            case tuya.dataPoints.ksmtMaxTemp:
                return {max_temperature: calibrateAndPrecisionRoundOptions(value / 10, options, 'temperature')};
            case tuya.dataPoints.ksmtMinTemp:
                return {min_temperature: calibrateAndPrecisionRoundOptions(value / 10, options, 'temperature')};
            case tuya.dataPoints.ksmtMaxHumidity:
                return {max_humidity: calibrateAndPrecisionRoundOptions(value, options, 'Humidity')};
            case tuya.dataPoints.ksmtMinHumidity:
                return {min_humidity: calibrateAndPrecisionRoundOptions(value, options, 'Humidity')};
            case tuya.dataPoints.ksmtTempAlarm:
                return {temperature_alarm: {0x00: 'canceled', 0x01: 'lower_alarm', 0x02: 'upper_alarm'}[value]};
            case tuya.dataPoints.ksmtHumidityAlarm:
                return {humidity_alarm: {0x00: 'canceled', 0x01: 'lower_alarm', 0x02: 'upper_alarm'}[value]};
            case tuya.dataPoints.ksmtTempSensitivity:
                return {temperature_sensitivity: calibrateAndPrecisionRoundOptions(value / 10, options, 'temperature')};
            case tuya.dataPoints.ksmtHumiditySensitivity:
                return {humidity_sensitivity: calibrateAndPrecisionRoundOptions(value, options, 'humidity')};
            case tuya.dataPoints.ksmtTempReport:
                return {temperature_report: calibrateAndPrecisionRoundOptions(value, options, 'value')};
[... 85 lines cut ...]
            .withValueMin(-20).withValueMax(60)
            .withDescription('Alarm temperature min'),
        exposes.numeric('max_humidity', ea.STATE_SET)
            .withUnit('%').withValueMin(0).withValueMax(100)
            .withDescription('Alarm humidity max'),
        exposes.numeric('min_humidity', ea.STATE_SET)
            .withUnit('%').withValueMin(0).withValueMax(100)
            .withDescription('Alarm humidity min'),
        exposes.numeric('temperature_sensitivity', ea.STATE_SET)
            .withUnit('°C').withValueMin(0.3).withValueMax(1).withValueStep(0.1)
            .withDescription('Temperature sensitivity'),
        exposes.numeric('humidity_sensitivity', ea.STATE_SET)
            .withUnit('%').withValueMin(3).withValueMax(10).withValueStep(1)
            .withDescription('Temperature sensitivity'),
        exposes.numeric('temperature_report', ea.STATE_SET)
            .withUnit('Seconds').withValueMin(1).withValueMax(120).withValueStep(1)
            .withDescription('Temperature periodic report'),
        exposes.numeric('humidity_report', ea.STATE_SET)
            .withUnit('Seconds').withValueMin(1).withValueMax(120).withValueStep(1)
            .withDescription('Humidity periodic report'),
    ],
};

module.exports = definition;
```

closed 2022-06-04 by github-actions[bot]

lowerprofile, 2022-06-22:
Everything I've done works fine, except configuring the device via the zigbee2mqtt web interface. Since I have a Tuya gateway I just used that to configure them and left it at that, since you only do it once. TBH I haven't even got around to trying via the command line.

owangen, 2022-06-23:
Just created a pull request for this device now https://github.com/Koenkk/zigbee-herdsman-converters/pull/4375

owangen, 2022-07-06:
> Everything I've done works fine, except configuring the device via the zigbee2mqtt web interface. Since I have a Tuya gateway I just used that to configure them and left it at that, since you only do it once. TBH I haven't even got around to trying via the command line.

I have tested it some more and notice some strange behavior. When you say configure the device, do you mean setting up reporting interval and such? I'm not able to send commands to it, meaning the values does not "stick" to the device. When I force a refresh of the device (pressing the button on the top) the old values are send to z2m. I also have the Tuya gateway and everything works fine what paired to that gateway.

lowerprofile, 2022-07-08:
Yes, that's what I meant. In the end I used the tuya gateway to set the
device up. I don't use the alarms, so once I've got the units and intervals
ser up I don't need to write to it any more.

owangen, 2022-07-08:
BTW, it's supported from next version because since with was supported from someone in this PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/4415, as you can see it only supports basic functionality like getting temperature and humidity, so it's not a full implementation of the device capability.