Koenkk/zigbee2mqtt #12286: [New device support]: GS SGPHM-I1

baradetsky, 2022-04-26:
### Link

https://www.gs.ru/catalog/umnyy-dom/datchik-gaza-propan-gs-sgphm-i1/

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0x000d6f0010dd05f0","nwkAddr":25071,"manufId":4619,"manufName":"GS","powerSource":"Mains (single phase)","modelId":"SGPHM-I1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,9,1280,65522],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0021b49930","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":17,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.26","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1650991985768,"defaultSendRequestWhen":"immediate"}

### Comments

Please add propane gas sensor GS sgphm-i1, coz its clone of HEIMAN Combustible gas sensor

baradetsky, 2022-05-06:
@Koenkk can u help me?

Koenkk, 2022-05-06:
Try with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SGPHM-I1'],
    model: 'SGPHM-I1',
    vendor: 'HEIMAN',
    description: 'Propane gas sensor',
    fromZigbee: [fz.ias_gas_alarm_1],
    toZigbee: [],
    exposes: [e.gas(), e.battery_low(), e.tamper()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if device works

baradetsky, 2022-05-12:
> * start z2m, check if device works
add code to zigbee2mqtt\configuration.yaml - DONE
make ext_converter.js with the code - DONE  
put ext_converter.js at the same directory - DONE
reboot - DONE
and and it didn't works(((
and now i can't delete it......

LOG:
Error 2022-05-12 20:30:28Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device '0x000d6f0010dd05f0' (block: false, force: false) (YAMLException: duplicated mapping key (50:1) 47 | port: 8099 48 | experimental: {} 49 | availability: false 50 | external_converters: ------^ 51 | - ext_converter.js)'
Warning 2022-05-12 20:30:29Device '0x000d6f0010dd05f0' left the network
Warning 2022-05-12 20:30:39Device '0x000d6f0010dd05f0' with Zigbee model 'SGPHM-I1' and manufacturer name 'GS' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

baradetsky, 2022-05-13:
> Can you provide me your complete `configuration.yaml`? (without any sensitive data like urls,password,usernames)

ОК
```
external_converters:
  - ext_converter.js
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: addons
  password: ************
  keepalive: 60
  reject_unauthorized: true
  version: 4
  include_device_information: true
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: warn
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: false

```

baradetsky, 2022-05-14:
WOW! It works!!! TNX!

closed 2022-05-14 by baradetsky

marselboss, 2024-01-09:
@Koenkk hello
I have the same problem: the sensor is online but does not respond. After setting up the external convector, everything works, but the sensor is always offline. It works, responds to the test, but is offline.