Koenkk/zigbee2mqtt #12292: HGZB-04D Does not report on/off changes

Adam-Langley, 2022-04-27:
### Link

https://www.ebay.com.au/itm/ZigBee-Smart-1-Gang-Trailing-Edge-Dimmer-Switch-EchoPlus-Hue-SmartThings-Hubitat/184686987635

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0x588e81fffe98934f","nwkAddr":44811,"manufId":4478,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN60-DS27LX1.3","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LXN60-DS27LX1.3","manufacturerName":"3A Smart Home DE","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20201106"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20201106","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1651011414037,"defaultSendRequestWhen":"immediate"}

### Comments

I have paired this switch multiple times and it never reports on/off or dimmer levels (proactively).
I can force a "read" of these values from within z2m however - that works.

I have another model of this dimmer switch, which is a 2-gang, and it pairs and reports correctly. I note that it has an extra in-cluster of "4096"

I dont know if that information helps you...

{"id":9,"type":"Router","ieeeAddr":"0x00158d00060f2b50","nwkAddr":53459,"manufId":4478,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN56-DS27LX1.3","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,3,6,8,5,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LXN56-DS27LX1.3","manufacturerName":"3A Smart Home DE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20191106","swBuildId":"1000-0001"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20191106","swBuildId":"1000-0001","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1651019277735,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-04-27:
Pushed the fix, can you check if it works?
1. update to latest dev
2. disable the configuring you setup manually
3. on the about page of the device reconfigure the device by pressing the yellow refresh button
4. see if state changes are reported now.
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Adam-Langley, 2022-04-27:
Thanks @Koenkk 
I'm running HomeAssistant on Pi (HAOS) so I assume I use the "Edge" addon?
How can I do this without destroying my current z2m installation and device registrations?
Thanks

Adam-Langley, 2022-04-27:
Hi Koenkk,

Thanks for turning this around so quickly.

I installed the edge version this morning (NZT), but saw no difference in behavior.
I dont see any mention in the changelog about this fix, so I'm not sure that I have the latest edge version.
[image]

How do I ensure I get the latest edge code? Following the instructions in the edge documentation, it says:
`Refresh repo: Supervisor → Add-on store → ⋮ → Reload`

There is no "reload" button, so not sure what it means.

Thanks,

Adam-Langley, 2022-04-28:
Hi @Koenkk - I've done everything I can think of to try to refresh the edge addon - including fully removing the repository and adding it back.
You mention the changelog doesnt look as expected? Looking at git, that changelog hasn't changed in 2 years - so are you expecting it to be different?
Just to remind you - I am using the hassio addon - https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/tree/master/zigbee2mqtt-edge

Anyway, the upshot is - I don't know how to tell what build of edge I'm running, hence cannot tell what source code it came from, hence cannot confirm if your fix is in my install. I *suspect* is has updated however, as I am getting *different* behavior.

I have performed the instructions you said, and there is a difference - after "reconfiguring" the light switch, the "Bind" is populated with the following:
[image]

However, the issue remains - the state does not update when the switch is manually manipulated.

I note there is nothing listed under "reporting" (I confess I do not know the purpose of the reporting tab) and if I can correct the issue with the following steps
1. unbind the onOff/LevelCtrl binding
2. add reporting entries for onOff and LevelCtrl (make sure you use "1" as the min rep interval)

These steps also result in a similar "bind" entry being added (I dont know why).
Now it all works.

How do we simulate these steps in the configure code/converter?

BTW: Thanks for bringing your attention to this issue - your help is definitely appreciated, even if my stumbling around doesn't contribute much in the way of diagnostics :)

Koenkk, 2022-04-29:
Ah you are still using the old repo:

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

Adam-Langley, 2022-04-29:
@Koenkk - sorry, I dont know what you are asking me to do.
This is my repo configuration. If that's not the URL I should have for z2m hassio, please let me know the correct URL.

[image]

Koenkk, 2022-04-30:
It is the old repo indeed, new one: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#installation

Adam-Langley, 2022-04-30:
Hi @Koenkk - color me confused - you're pointing me to the same repo I already have installed... what am I missing?

[image]

Koenkk, 2022-05-01:
Ah sorry, I was confused, you are using the correct repo. You can also check the 1.25.1-1 version which has been released just now (should show up in an hour from now)

closed 2022-06-09 by github-actions[bot]