Koenkk/zigbee2mqtt #12296: Aqara Opple switch 3 bands

Jens-Wymeersch, 2022-04-27:
### What happened?

I just bought some Aqara Opple switches (https://www.zigbee2mqtt.io/devices/WXCJKG13LM.html)

### What did you expect to happen?

I have already 2 switches running smoothly but the new ones I just bought aren't reacting.

1.  I don't see any battery information
2. I don't see action - when I push one of the buttons

making the product useless.

[image]

### How to reproduce it (minimal and precise)

I bought 4 devices and all have the same problem.

### Zigbee2MQTT version

1.25.0 

### Adapter firmware version

20220103

### Adapter

ZZH

### Debug log

`Error Failed to configure '0x54ef441000456366', attempt 1 (Error: Write 0x54ef441000456366/1 aqaraOpple({"mode":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 19505 - 1 - 150 - 64704 - 4 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))`

Jens-Wymeersch, 2022-10-18:
This is the firmware of the zzh stick. So followed the normal procedure as described in z2m documentation

wuppiwuppi, 2022-10-18:
Oh, this does not help then. I have a RaspBee 1 and your mentioned battery issue. 
But it does receive actions (single_press, hold, release)

Coordinator Typ ConBee/RaspBee
Coordinator Version 0x26390500

wuppiwuppi, 2022-10-20:
removed from z2m
joined
joined a 2nd time
works now ...