Koenkk/zigbee2mqtt #12299: [New device support]: MoesGo EDM-1ZBB-EU

timota, 2022-04-27:
### Link

https://www.moeshouse.com/en-gb/products/new-zigbee-smart-touch-light-dimmer-switch-timer-brightness-memory-smart-life-tuya-app-remote-control-work-with-alexa-google?_pos=11&_sid=64e9f8bed&_ss=r

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0x60a423fffe61bfe5","nwkAddr":29506,"manufId":4098,"manufName":"_TZE200_dfxkcots","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_dfxkcots","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca1bda7","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca1bda7","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1,"reporting":1},"lastSeen":1651090472323,"defaultSendRequestWhen":"immediate"}

### Comments

Actual device is MoesGo `EDM-1ZBB-EU`. (this device doesnt exists in devices database, closes device is: EDM-1ZAB-EU)
Appear in web ui as: 
```
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_dfxkcots
```

When trying to manipulate by device (via UI or manually) - like changing on/off state or brightness(MOSTLY) in logs, such an error

```
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,228]}}': type='commandDataResponse', datatype='value', value='740', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,43]}}': type='commandDataReport', datatype='value', value='43', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,75]}}': type='commandDataReport', datatype='value', value='331', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
```

To be honest i don't know if this device needs to be added as a new device or this is bug in converters which is only related to current model. 

Please help. Thanks

timota, 2022-04-28:
any help please

timota, 2022-04-28:
Closed. Report has been created instead

closed 2022-04-28 by timota

Zenthae, 2022-05-20:
i have the same issue but with another device, have you been able to fix your issue ?