Koenkk/zigbee2mqtt #12300: [New device support]: TM616EGZT-V2version12V

samiracho, 2022-04-27:
### Link

https://es.aliexpress.com/item/1005003611118732.html?spm=a2g0o.productlist.0.0.453334c33f7c4L&algo_pvid=8b2b3e48-d2bb-4ccb-9d33-e4776a50b853&algo_exp_id=8b2b3e48-d2bb-4ccb-9d33-e4776a50b853-8&pdp_ext_f=%7B%22sku_id%22%3A%2212000026485305868%22%7D&pdp_npi=1%40dis%7CEUR%7C%7C21.17%7C%7C%7C1.5%7C%7C%400b0a182b16510936354436174e0711%7C12000026485305868%7Csea

### Database entry

{ 	"id": 5, 	"type": "Router", 	"ieeeAddr": "0x847127fffe0a8854", 	"nwkAddr": 43356, 	"manufId": 4098, 	"manufName": "_TZE200_hsgrhjpf", 	"powerSource": "Mains (single phase)", 	"modelId": "TS0601", 	"epList": [1], 	"endpoints": { 		"1": { 			"profId": 260, 			"epId": 1, 			"devId": 81, 			"inClusterList": [0, 4, 5, 61184], 			"outClusterList": [25, 10], 			"clusters": { 				"genBasic": { 					"attributes": { 						"modelId": "TS0601", 						"manufacturerName": "_TZE200_hsgrhjpf", 						"powerSource": 1, 						"zclVersion": 3, 						"appVersion": 66, 						"stackVersion": 0, 						"hwVersion": 1, 						"dateCode": "" 					} 				} 			}, 			"binds": [], 			"configuredReportings": [], 			"meta": {} 		} 	}, 	"appVersion": 66, 	"stackVersion": 0, 	"hwVersion": 1, 	"dateCode": "", 	"zclVersion": 3, 	"interviewCompleted": true, 	"meta": {}, 	"lastSeen": 1650623617098, 	"defaultSendRequestWhen": "immediate" }

### Comments

This new version doesn't allow to set the upper and lower limits with the rf controller or the physical buttons. According to the instructions he limits have to be set with app.
Following the instructions on how to support new tuya devices I made this converter, and after many trial error I was able to set up the upper and lower limits with the exposed options border down, up, down_delete

This model _TZE200_hsgrhjpf appears as supported, but without the custom converter I'm unable to set the limits. Motor speed and motor direction are not applicable to this model.

### External converter

```shell
const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee' );
const tz = require( 'zigbee-herdsman-converters/converters/toZigbee' );
const exposes = require( 'zigbee-herdsman-converters/lib/exposes' );
const reporting = require( 'zigbee-herdsman-converters/lib/reporting' );
const extend = require( 'zigbee-herdsman-converters/lib/extend' );
const tuya = require( 'zigbee-herdsman-converters/lib/tuya' );
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja',
            'mcdj3aq', 'mcdj3aq\u0000',
            ],
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE200_hsgrhjpf' },
            ],           
        model: 'TS0601_cover',
        vendor: 'Tuya',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Tuya', model: 'TM616EGZT'},
            ],
        fromZigbee: [
            fz.ZMAM02_cover
            ],
        toZigbee: [
            tz.ZMAM02_cover,
            ],
        onEvent: tuya.onEventSetTime,
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.enum('border', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Border))
            ],
    };

module.exports = definition;
```

samirachosentryc, 2022-05-21:
I think is the exposes.enum('border...
If I'm not wrong I had to click on border_delete, then move the motor up and click on up and finally move the motor down and click on down.
It took me many tries.
[image]

mrJudahBella, 2022-06-11:
Following #11594 I managed to get motor direction to work by adding 

```
exposes.enum('motor_direction', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Direction)), // forward/backward
```
to
```
exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.enum('motor_direction', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Direction)), // forward/backward
            exposes.enum('border', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Border)),
            ],
```

`border` `down_delete` is weird tho, it only deletes the `up` border. Not a big deal, I can still make it work.

closed 2022-07-20 by github-actions[bot]

fluffyanvil, 2022-09-20:
up