Koenkk/zigbee2mqtt #12301: Aqara WS-EUK03 sends "ON" to Zigbee2mqtt , Zigbee2mqtt broadcasts OFF

Multipage, 2022-04-28:
### What happened?

**Situation**
Bought 6 Xiaomi Aqara WS-EUK3
I Have a ceiling with a Sonoff iFan03 (flashed with Tasmota)
The WS-EUK03 is **NOT** physically connected to the Sonoff/Ceiling fan/light. The only wires connected to the WS-EUK03 are NULL(neutral) and LIVE, L1 is I( like i said ) **NOT** connected to the ceiling fan/light.
Made and automation that turns the ceiling fan/light on/off based on the state of the entity of the WS-EUK03

Problem:
Sometimes the wrong message is published by Zigbee2MQTT and autoomation won't start of this

### What did you expect to happen?

Turn on ceiling fan/light when WS-EUK03 is turned on, Turn off ceiling fan/light when WS-EUK03 is turned off.

**NOTE!!!!**
If you connect the L1 line to a bulb you THINK it works fine because the light turns on: the physical connection works, that's not the problem.
But if you look into Zigbee2MQTT of HomeAssitant the state of the WS-EUK03 is OFF and that's offcourse wrong.

To fix this i think Zigbee2MQTT will only need to do something with the second message it recieves 
`Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 41 with groupID 0`

Or ALSO use it to set the state of the entity of the WS-EUK03

### How to reproduce it (minimal and precise)

1) Get an Aqara WS-EUK03 and only connect the Null(Neutral) and Live line, do NOT connect the L1 connection
2) Now push the WS-EUK03 button to on and off. Zigbee2MQTT will fail very quickly to publish the right Z2M message
- receives present state 1, but publishes OFF. 
- Or the other way around: receives present state 0, but publishes ON.

What i can see and uinderstand from the logfile is that there is 1 message missing if the state fails.
If it's working Z2M recieves 2 messages form the Aqara:
`Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genOnOff', data '{"245":57654784,"onOff":1}' from endpoint 1 with groupID 0`
and 
`Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 41 with groupID 0`

If it fails it only recieves 1 :
`Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 41 with groupID 0`

What i think that needs to be done to get this working right is instead of using the first

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

When it **doesn't** turn on ( **Unsuccesfully** set entity state to ON)
`Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 41 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer', payload '{"action":"single","consumption":0,"current":0,"energy":0,"flip_indicator_light":"ON","illuminance":0,"illuminance_lux":0,"linkquality":105,"operation_mode":null,"power":0,"power_outage_count":2,"power_outage_memory":null,"state":"OFF","temperature":29,"voltage":226.8}'
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer', payload '{"action":"","consumption":0,"current":0,"energy":0,"flip_indicator_light":"ON","illuminance":0,"illuminance_lux":0,"linkquality":105,"operation_mode":null,"power":0,"power_outage_count":2,"power_outage_memory":null,"state":"OFF","temperature":29,"voltage":226.8}'
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer/action', payload 'single'` 

When it **does** turn on When it **doesn't** turn on ( Succesfully set entity state to ON):
`Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genOnOff', data '{"245":57654784,"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer', payload '{"action":null,"consumption":0,"current":0,"energy":0,"flip_indicator_light":"ON","illuminance":0,"illuminance_lux":0,"linkquality":91,"operation_mode":null,"power":0,"power_outage_count":2,"power_outage_memory":null,"state":"ON","temperature":29,"voltage":227.2}'
Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 41 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer', payload '{"action":"single","consumption":0,"current":0,"energy":0,"flip_indicator_light":"ON","illuminance":0,"illuminance_lux":0,"linkquality":116,"operation_mode":null,"power":0,"power_outage_count":2,"power_outage_memory":null,"state":"ON","temperature":29,"voltage":227.2}'
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer', payload '{"action":"","consumption":0,"current":0,"energy":0,"flip_indicator_light":"ON","illuminance":0,"illuminance_lux":0,"linkquality":116,"operation_mode":null,"power":0,"power_outage_count":2,"power_outage_memory":null,"state":"ON","temperature":29,"voltage":227.2}'
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer/action', payload 'single'`

When it **doesn't ** turn off:
Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 21 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer', payload '{"action":null,"consumption":0,"current":0,"energy":0,"flip_indicator_light":"ON","illuminance":0,"illuminance_lux":0,"linkquality":116,"operation_mode":null,"power":0,"power_outage_count":2,"power_outage_memory":null,"state":"OFF","temperature":29,"voltage":227.2}'`

When it **does** turn off ( Succesfully set entity state to OFF):
`Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genOnOff', data '{"245":57654784,"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer', payload '{"action":null,"consumption":0,"current":0,"energy":0,"flip_indicator_light":"ON","illuminance":0,"illuminance_lux":0,"linkquality":98,"operation_mode":null,"power":0,"power_outage_count":2,"power_outage_memory":null,"state":"OFF","temperature":29,"voltage":227.2}'
Debug Received Zigbee message from 'Licht slaapkamer', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 21 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Licht slaapkamer', payload '{"action":null,"consumption":0,"current":0,"energy":0,"flip_indicator_light":"ON","illuminance":0,"illuminance_lux":0,"linkquality":116,"operation_mode":null,"power":0,"power_outage_count":2,"power_outage_memory":null,"state":"OFF","temperature":29,"voltage":227.2}'`

Multipage, 2022-04-28:
Update: Experiencing the same Problem with the Xiaomi Aqara WS-EUK01 and WS-EUK02. I saw the bathroomlight reported it was ON (WS-EUK02). Checked it ...it was physically turned OFF. 
Walked to the pantry(WS-EUK01)..turned light on and off ...and yes ..same result but in this case light was ON physically , switch was physically on but Z2M andm HomeAssistant reported it to be OFF.

If i turn on/off the lights i'm not doing it like a mad men. Just slowly.

McGiverGim, 2022-05-03:
Let's start with the WS-EUK03.

Looking at the code an your debug:
- z2m turns "on" and "off" only when receives the `genOnOff` cluster. 
- The `genMultistateInput` is received when you press the button, to indicate the `single` pulsation. But it seems it does not imply the on/off of the device.
- The `genAnalogInput` is used to send the power comsumption.
- There is other cluster that can be received `aqara_opple`, this is usually send by the device to "update" some of the attributes, specially the power for example, periodically.

Without other messages, z2m does not seem to be able to fix the issue, at least processing the messages received. It seems the `genOnOff` message is missing sometimes.

closed 2022-06-11 by github-actions[bot]