Koenkk/zigbee2mqtt #12302: Müller Remote 404049 gets detected as 404022

DevFlorian, 2022-04-28:
### What happened?

Remote 404049 gets detected as 404022, but should bei detected as 404011 to support colorwheel

https://www.mueller-licht.de/produktinformationen/artikel/?q=404049
https://www.mueller-licht.de/produktinformationen/artikel/?q=404022
https://www.mueller-licht.de/produktinformationen/artikel/?q=404011

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

6.7.8.0 build 373

### Adapter

EZSP v8

Koenkk, 2022-05-03:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4194

closed 2022-05-03 by Koenkk

Koenkk, 2022-07-05:
@ArusHear can you provide the data/database.db entry of your device?

ArusHear, 2022-07-07:
@Koenkk of course 
`{"id":32,"type":"EndDevice","ieeeAddr":"0xa4c13800aa603bd8","nwkAddr":18172,"manufId":4635,"manufName":"MLI","powerSource":"Battery","modelId":"tint-Remote-white","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2048,"inClusterList":[0,3],"outClusterList":[3,4,6,8,25,4096,4111],"clusters":{"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"MRM001","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1657029748835,"defaultSendRequestWhen":"immediate"}`

ArusHear, 2022-07-07:
@Koenkk Yes, my model is 4040449, let see
[image]
[image]

Twestus, 2022-09-20:
Hi,
I have the same issue. Both of my tint 404049C are detected as 404022.
So I miss for example the color wheel and the scene buttons.
Is there any solution?
Thanks