Koenkk/zigbee2mqtt #12312: Z2M GUI not shown in HomeAssistant iOS App

michalk-k, 2022-04-28:
### What happened?

Recently I updated HA from 2021.12 to 2022.4 (through the most recent 2022.3) together with Zigbee2MQTT addon to version 1.25.0-1. Everything works as expected, incl GUI running inside a desktop browser (Chrome, Firefox).

However, when trying to open Z2M gui from within HA iOS app (official), it opens empty content instead Z2M GUI

I don't know if it is an issue with Z2M or HA.

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

0x26700700

### Adapter

ConBee2

michalk-k, 2022-04-30:
just fyi: updating HA to most 2022.4.7 didn't help

MJWMJW2, 2022-05-01:
I just had the same symptom after updating Z2M to 1.25.1-1.  It was opening ok on iPad, but not on Mac, or iPhone.  Mac Safari - cleared browser cache.  IOS app I went to Configuration, Companion App, Debugging, Reset frontend cache.  Now it is working again for me.  I hope this tip may work for you also?

Aldaran, 2022-05-01:
> I just had the same symptom after updating Z2M to 1.25.1-1. It was opening ok on iPad, but not on Mac, or iPhone. Mac Safari - cleared browser cache. IOS app I went to Configuration, Companion App, [...]

On one server, I also cleared the cache and it helped, on the other it didn’t help ... Also, there are no logs like "Zigbee2MQTT: info 2022-05-02 00:17:09: MQTT publish: topic...." in the log, but the device still works...

TheRealWaldo, 2022-05-03:
Same problem here.

  Multiple tablets, phones, and Macs. iOS and Android clients.  On Chrome, the header menu disappears sometimes and only a full cache clear and reload fixes it temporarily.

Haven't been able to get it to show at all in the apps.

closed 2022-06-11 by github-actions[bot]