Koenkk/zigbee2mqtt #12313: MoesGo EDM-1ZBB-EU reported "Received unexpected Tuya DataPoint"

timota, 2022-04-28:
### What happened?

Device is `MoesGo EDM-1ZBB-EU`. 
Url: https://www.moeshouse.com/en-gb/products/new-zigbee-smart-touch-light-dimmer-switch-timer-brightness-memory-smart-life-tuya-app-remote-control-work-with-alexa-google?_pos=11&_sid=64e9f8bed&_ss=r

identified by Z2M as:
```
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_dfxkcots
```
When trying to manipulate by device (via UI or manually) - changing on/off state or brightness(MOSTLY) in logs i see such an errors(below). After bunch of errors - device become unresponsive to Z2M commands and reboot required. 

Release
```
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,228]}}': type='commandDataResponse', datatype='value', value='740', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,43]}}': type='commandDataReport', datatype='value', value='43', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,75]}}': type='commandDataReport', datatype='value', value='331', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
```
Dev:
```
Warning 2022-04-28 21:29:45zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,3,161]}}': type='commandDataReport', datatype='value', value='929', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","connecteMode"]
Warning 2022-04-28 21:30:33zigbee-herdsman-converters:tuya_dimmer: Received Tuya DataPoint #2 with invalid value 0 for brightness from 0x60a423fffe61bfe5 which is outside the expected range from 10 to 1000
```

Have looked around other issues and found this one: https://github.com/Koenkk/zigbee-herdsman-converters/pull/3004

For me looks like an issue with this part of code: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L4764-L4775 

But, unfortunately im not familiar with ZigBee too much, so can not identify actual issue and what needs to be changed. 

Any help please

### What did you expect to happen?

Errors will gone and devices will behave as it should.

### How to reproduce it (minimal and precise)

Via UI. - switch on/off. Change brightness level.

### Zigbee2MQTT version

1.25.0-dev

### Adapter firmware version

latest available

### Adapter

Texas Instruments LAUNCHXL-CC1352P-2

Koenkk, 2022-04-29:
to figure out what this data points means follow: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

timota, 2022-04-29:
Thanks for advice. 
Btw, i forget to mention that device working as expected(even if i see error in logs), until flood of logs comes. 

Reconnected device to GW and got following DPs
```
switch      1
brightness  2
minimum brightness  3
type of light source    4
Indicator light mode    21
mode    13
scene  101
```

also,  did more test and see that those errors coming when using brightness. And im still believe that issue somewhere here 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L4764-L4775. 

@Koenkk could you please with next steps. 
Thanks in advance.

Koenkk, 2022-04-30:
Should be fixed now!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

timota, 2022-04-30:
Yeo, issue fixed. Thank you

timota, 2022-04-30:
Device was fixed, all operations works as expected. But with some time, like 10-20 mins - device starts loosing network and become "dead". Some logs.

```
Debug 2022-04-30 21:21:47Received MQTT message on 'zigbee2mqtt/0x60a423fffe61bfe5/set' with data '{"state":"OFF"}'
Debug 2022-04-30 21:21:47Publishing 'set' 'state' to '0x60a423fffe61bfe5'
Debug 2022-04-30 21:21:48Received MQTT message on 'zigbee2mqtt/0x60a423fffe61bfe5/set' with data '{"state":"OFF"}'
Debug 2022-04-30 21:21:48Publishing 'set' 'state' to '0x60a423fffe61bfe5'
Debug 2022-04-30 21:21:50Received MQTT message on 'zigbee2mqtt/0x60a423fffe61bfe5/set' with data '{"brightness":88}'
Debug 2022-04-30 21:21:50Publishing 'set' 'brightness' to '0x60a423fffe61bfe5'
Error 2022-04-30 21:21:51Publish 'set' 'state' to '0x60a423fffe61bfe5' failed: 'Error: Command 0x60a423fffe61bfe5/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
Debug 2022-04-30 21:21:51Error: Command 0x60a423fffe61bfe5/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Info 2022-04-30 21:21:51MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x60a423fffe61bfe5' failed: 'Error: Command 0x60a423fffe61bfe5/1 manuSpecificTuya.dataRequest({\"seq\":1,\"dpValues\":[{\"dp\":1,\"datatype\":1,\"data\":[0]}]}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"0x60a423fffe61bfe5"},"type":"zigbee_publish_error"}'
Error 2022-04-30 21:22:01Publish 'set' 'state' to '0x60a423fffe61bfe5' failed: 'Error: Command 0x60a423fffe61bfe5/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
Debug 2022-04-30 21:22:01Error: Command 0x60a423fffe61bfe5/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Info 2022-04-30 21:22:01MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x60a423fffe61bfe5' failed: 'Error: Command 0x60a423fffe61bfe5/1 manuSpecificTuya.dataRequest({\"seq\":1,\"dpValues\":[{\"dp\":1,\"datatype\":1,\"data\":[0]}]}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"0x60a423fffe61bfe5"},"type":"zigbee_publish_error"}'
Error 2022-04-30 21:22:10Publish 'set' 'state' to '0x60a423fffe61bfe5' failed: 'Error: Command 0x60a423fffe61bfe5/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
Debug 2022-04-30 21:22:10Error: Command 0x60a423fffe61bfe5/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request  [... cut ...] :1,\"data\":[0]}]}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"0x60a423fffe61bfe5"},"type":"zigbee_publish_error"}'
```

is it something with a device ? or can be fixed ?

timota, 2022-04-30:
Some other behaviour, during writing this post - device restored itself somehow. 
```
Debug 2022-04-30 21:28:54Device '0x60a423fffe61bfe5' reconnected
```

Koenkk, 2022-05-02:
Mac channel access failure is caused by interference: https://www.zigbee2mqtt.io/guide/faq/#common-error-codes

timota, 2022-05-05:
Yes, definitely it was an interference issue. Somehow my WIFI router after upgrade switched to channel 1 which is overlapped with zigbee ch11. 
All your changes above works as expected. tested on 3 dimmers. 
Many thanks.

closed 2022-05-05 by timota