Koenkk/zigbee2mqtt #12315: [New device support]: Saswell SEA801-Zigbee/SEA802-Zigbee, Additional manufacturer name

SimonSezKossel, 2022-04-29:
### Link

https://nl.aliexpress.com/item/1005003952150617.html?spm=a2g0o.order_list.0.0.773979d2HwK8Cy&gatewayAdapt=glo2nld

### Database entry

{"id":49,"type":"EndDevice","ieeeAddr":"0x0c4314fffe61c76e","nwkAddr":39148,"manufId":4098,"manufName":"_TZE200_bvu2wnxz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"!��)\u0012\"��)\u0012\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0011","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_bvu2wnxz","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1651227141902,"defaultSendRequestWhen":"immediate"}

### Comments

Bought a extra Thermostatic radiator valve. Expected that it would be compatible with zigbee2mqtt right out of the box. Unfortunately it is not. But I am assuming that it is identical to the Saswell SEA801 functionality. 
Can we please add a new model id to the saswell.js fingerprint list, so I can check if it works:
        **{modelID: 'TS0601', manufacturerName: '_TZE200_bvu2wnxz'}**

### External converter

```shell
Tried a modified saswell.js as external converter, but I got a lot of errors.
```

Koenkk, 2022-04-29:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and next release on 1 May.

closed 2022-04-29 by Koenkk

SimonSezKossel, 2022-04-30:
Thank you Koen, for the quick fix.

Unfortunately it appears not to be as simple as I hoped. This device seems not to follow the standard saswell comunication.
Getting a lot of warnings with Unrecognized DP #numbers

What do you suggest to do next? I am willing to help but my programming skills are none existing.

Log:
Info 2022-04-30 10:46:55Successfully interviewed '0x0c4314fffe61c76e', device has successfully been paired
Info 2022-04-30 10:46:55Device '0x0c4314fffe61c76e' is supported, identified as: Saswell Thermostatic radiator valve (SEA801-Zigbee/SEA802-Zigbee)
Info 2022-04-30 10:46:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Thermostatic radiator valve","exposes":[{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"features":[{"access":3,"description":"Enables/disables window detection on the device","name":"state","property":"window_detection","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":1,"description":"Device valve is open or closed (heating or not)","name":"heating","property":"heating","type":"binary","value_off":"OFF","value_on":"ON"},{"features":[{"access":3,"description":"Temperature setpoint","name":"current_heating_setpoint","property":"current_heating_setpoint","type":"numeric","unit":"°C","value_max":30,"value_min":5,"value_step":0.5},{"access":1,"description":"Current temperature measured on the device","name":"local_temperature","property":"local_temperature","type":"numeric","unit":"°C"},{"access":3,"description":"Mode of this device","name":"system_mode","property":"system_mode","type":"enum","values":["off","heat","auto"]},{"access":3,"description":"Offset to be used in the local_temperature","name":"local_temperature_calibration","property":"local_temperature_calibration","type":"numeric","unit":"°C","value_max":6,"value_min":-6,"value_step":1},{"access":3,"description":"Away mode","name":"away_mode","property":"away_mode","type":"binary","value_off":"OFF","value_on":"ON"}],"type":"climate"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SEA801-Zigbee/SEA802-Zigbee","options":[{"access":2,"description":"Set to false to disable the legacy integration (highly recommended), will change structure of the published payload (default true).","name":"legacy","property":"legacy","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Saswell"},"friendly_name":"0x0c4314fffe61c76e","ieee_address":"0x0c4314fffe61c76e","status":"successful","supported":true},"type":"device_interview"}'
Info 2022-04-30 10:46:55Configuring '0x0c4314fffe61c76e'
Info 2022-04-30 10:46:55MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Thermostatic radiator valve","friendly_name":"0x0c4314fffe61c76e","model":"SEA801-Zigbee/SEA802-Zigbee","supported":true,"vendor":"Saswell"},"type":"pairing"}'
Info 2022-04-30 10:46:55Successfully configured '0x0c4314fffe61c76e'
Debug 2022-04-30 10:46:57Received Zigbee message from '0x0c4314fffe61c76e', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":2}],"seq":49664}' from endpoint 1 with groupID 0
Warning 2022-04-30 10:46:57zigbee-herdsman-converters:SaswellThermostat: Unrecognized DP #2 with data {"dp":2,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Debug 2022-04-30 10:46:58Received Zigbee message from '0x0c4314fffe61c76e', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":3}],"seq":49920}' from endpoint 1 with groupID 0
Info 2022-04-30 10:46:58MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe61c76e', payload '{"battery_low":null,"child_lock":null,"heating":"ON","linkquality":69,"local_temperature_calibration":null,"window_detection":null}'
Debug 2022-04-30 10:46:58Received Zigbee message from '0x0c4314fffe61c76e', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,50],"type":"Buffer"},"datatype":2,"dp":4}],"seq":50176}' from endpoint 1 with groupID 0
Warning 2022-04-30 10:46:58zigbee-herdsman-converters:SaswellThermostat: Unrecognized DP #4 with data {"dp":4,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,50]}}
Debug 2022-04-30 10:46:58Received Zigbee message from '0x0c4314fffe61c76e', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,180],"type":"Buffer"},"datatype":2,"dp":5}],"seq":50432}' from endpoint 1 with groupID 0
Warning 2022-04-30 10:46:58zigbee-herdsman-converters:SaswellThermostat: Unrecognized DP #5 with data {"dp":5,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,180]}}
Debug 2022-04-30 10:46:58Received Zigbee message from '0x0c4314fffe61c76e', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":7}],"seq":50688}' from endpoint 1 with groupID 0
[... cut ...]
Debug 2022-04-30 10:46:59Received Zigbee message from '0x0c4314fffe61c76e', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":21761}' from endpoint 1 with groupID 0
Debug 2022-04-30 10:47:00Received Zigbee message from '0x0c4314fffe61c76e', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,180],"type":"Buffer"},"datatype":2,"dp":5}],"seq":54016}' from endpoint 1 with groupID 0
Warning 2022-04-30 10:47:00zigbee-herdsman-converters:SaswellThermostat: Unrecognized DP #5 with data {"dp":5,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,180]}}

xyMichaelxy, 2022-07-16:
> Thank you Koen, for the quick fix.
> 
> Unfortunately it appears not to be as simple as I hoped. This device seems not to follow the standard saswell comunication. Getting a lot of warnings with Unrecognized DP #numbers

I have the same problem!
Is there a solution for this now?

zigbee-herdsman-converters:SaswellThermostat: Unrecognized DP #5 with data {"dp":5,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,200]}}

This is one of the many messages I get when pairing the SAE802-Z01

codigospostales, 2022-08-09:
Same problem here.
Fortunatelly only one thermostat from this brand.
Waiting for a solution (if any).

martin19x, 2022-08-11:
I had ordered a different thermostat, but the AliExpress dealer sent me the same TVR {modelID: 'TS0601', manufacturerName: '_TZE200_bvu2wnxz'}.
On the package it says the type ME167.

Is there any solution for the problem?

CR-SilvioRiener, 2022-08-27:
I'd be very interested in a solution for this as well - got four of this valves from aliexpress :(

GrumpyMeow, 2022-09-19:
I'm in the same boat :-(

Slaughthammer, 2022-09-29:
Damn, should have seen this before ordering 5 of these... is there still hope for these functioning in foreseeable future?

Slaughthammer, 2022-09-30:
got some basic functions out of mine with help of this: https://github.com/twhittock/avatto_me167

https://github.com/Koenkk/zigbee2mqtt/issues/13727

ebyna, 2022-09-30:
Also a thermostatic valve.
Marking AVATTO ME167, not functional. If anyone can please add to basic support. Thank you

artist67, 2022-10-18:
I have the probably the same device now. branded Avatto.

DB:
{"id":136,"type":"EndDevice","ieeeAddr":"0x040d84fffe96df77","nwkAddr":50039,"manufId":4098,"manufName":"_TZE200_bvu2wnxz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{terList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"e��*fe��*\u0012e��*\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_bvrsion":65}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666123

It is recognized as "TuYa TV02-Zigbee"

Effectively, it does not work. See log:
info  2022-10-18 22:04:35: Device 'W_Bad' joined
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"W_Bad","ieee_address":"0x040d84fffe96df77"},"type":"device_joined"}'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/last_seen', payload '2022-10-18T20:04:36.033Z'
info  2022-10-18 22:04:36: Starting interview of 'W_Bad'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"W_Bad","ieee_address":"0x040d84fffe96df77","status":"started"},"type":"device_interview"}'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/linkquality', payload '174'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/last_seen', payload '2022-10-18T20:04:36.142Z'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"W_Bad","ieee_address":"0x040d84fffe96df77"},"type":"device_announce"}'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/linkquality', payload '174'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/last_seen', payload '2022-10-18T20:04:36.376Z'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/linkquality', payload '167'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/last_seen', payload '2022-10-18T20:04:36.830Z'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/linkquality', payload '163'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/last_seen', payload '2022-10-18T20:04:36.885Z'
info  2022-10-18 22:04:36: MQTT publish: topic 'zigbee2mqtt/W_Bad/linkquality', payload '167'
info  2022-10-18 22:04:37: MQTT publish: topic 'zigbee2mqtt/W_Bad/last_seen', payload '2022-10-18T20:04:37.178Z'
info  2022-10-18 22:04:37: MQTT publish: topic 'zigbee2mqtt/W_Bad/linkquality', payload '167'
info  2022-10-18 22:04:37: Successfully interviewed 'W_Bad', device has successfully been paired
info  2022-10-18 22:04:37: Device 'W_Bad' is supported, identified as: TuYa Thermostat radiator valve (TV02-Zigbee)
info  2022-10-18 22:04:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Thermostat radiator valve","exposes":[{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":3,"description":"Enables/disables the status on the device","name":"open_window","property":"open_window","type":"binary","value_off":"OFF","value_on":"ON"},{"access":3,"description":"Open window temperature","name":"open_window_temperature","property":"open_window_temperature","type":"numeric","unit":"°C","value_max":30,"value_min":0},{"access":3,"description":"Comfort temperature","name":"comfort_temperature","property":"comfort_temperature","type":"numeric","unit":"°C","value_max":30,"value_min":0},{"access":3,"description":"Eco temperature","name":"eco_temperature","property":"eco_temperature","type":"numeric","unit":"°C","value_max":30,"value_min":0},{"features":[{"access":3,"description":"Mode of this device (similar to system_mode)","name":"preset","property":"preset","type":"enum","values":["auto","manual","holiday"]},{"access":3,"description":"Offset to be used in the local_temperature","name":"local_temperature_calibration","property":"local_temperature_calibration","type":"numeric","unit":"°C","value_max":5,"value_min":-5,"value_step":0.1},{"access":1,"description":"Current temperature measured on the device","name":"local_temperature","property":"local_temperature","type":"numeric","unit":"°C"},{"access":3,"description":"Temperature setpoint","name":"current_heating_setpoint","property":"current_heating_setpoint","type":"numeric","unit":"°C","value_max":30,"value_min":0,"value_step":0.5}],"type":"climate"},{"access":3,"description":"Setting minimum 0 - maximum 465 seconds boost time. The boost (♨) function is activated. The remaining time for the function will be counted down in seconds ( 465 to 0 ).","name":"boost_timeset_countdown","property":"boost_timeset_countdown","type":"numeric","unit":"second","value_max":465,"value_min":0},{"access":3,"description":"When Anti-Freezing function is activated, the temperature in the house is kept at 8 °C, the device display \"AF\".press the pair button to cancel.","name":"frost_protection","property":"frost_protection","type":"binary","value_off":"OFF","value_on":"ON"},{"access":3,"description":"Battery life can be prolo [... cut ...]
info  2022-10-18 22:04:44: MQTT publish: topic 'zigbee2mqtt/W_Bad/last_seen', payload '2022-10-18T20:04:44.408Z'
info  2022-10-18 22:04:44: MQTT publish: topic 'zigbee2mqtt/W_Bad/linkquality', payload '171'
warn  2022-10-18 22:04:44: zigbee-herdsman-converters:tvtwo_thermostat: NOT RECOGNIZED DP #5 with data {"dp":5,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,190]}}
info  2022-10-18 22:04:44: MQTT publish: topic 'zigbee2mqtt/W_Bad/last_seen', payload '2022-10-18T20:04:44.877Z'
info  2022-10-18 22:04:44: MQTT publish: topic 'zigbee2mqtt/W_Bad/linkquality', payload '171'

tracetechnical, 2022-11-10:
Same issue here with KETOTEK KTF0177 from Amazon UK.

olexs, 2022-11-12:
Same issue for me. I have the "ME167" branded device. Looks absolutely identical to the Saswell 802, but does not work with an external converter with the Saswell definition, only throws errors. Also tested some other Tuya thermostat definitions, with no success so far.

artist67, 2022-11-12:
There is an external converter available (search github for me167).
It might be officially included in the future, but it handles only the most relevant attributes.

mikeg-de, 2022-11-17:
I tried that converter but it does not work. Even attempting to read the current temperature throws an error:

`No converter available for 'local_temperature' ("")`

Did anyone had success in getting that thermostat to work? My attempts in https://github.com/Koenkk/zigbee2mqtt/issues/15009 failed unfortunately.

Slaughthammer, 2022-11-17:
It works. It's just not possible to actively get the parameters, you have to wait ~10 minutes until the TRV sends the data on its own. Not perfect, but useable.

olexs, 2022-11-17:
Yes, the external "me167" converter works for me too. There is no battery readout and as @Slaughthammer says values cannot be actively read, but setting values and getting updates from the TRV works fine.

mikeg-de, 2022-11-17:
Was about to reply as after a few minutes data arrived and assume pulling didn't work even if the device is active. I noticed the battery icon in M2Z blinking (deliberately inserted an under-volted 1.2 V battery). Are you certain the battery status is not providing a readout? I prev. used 'e.battery()',

MartiPri, 2022-11-19:
The _TZE200_bvu2wnxz does not work. You added it to the dev ?😐

{"id":5,"type":"EndDevice","ieeeAddr":"0x84ba20fffe9a63a7","nwkAddr":13049,"manufId":4098,"manufName":"_TZE200_bvu2wnxz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":31,"65508":1,"appVersion":65,"modelId":"TS0601","manufacturerName":"_TZE200_bvu2wnxz","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1668882390722,"defaultSendRequestWhen":"immediate"}

> Added!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and next release on 1 May.

charly50, 2022-11-22:
Hi, i'm the next guy, how has the same issue with Ketotek _TZE200_bvu2wnxz :-( My HA Zigbee2MQTT shows "unsupported". Who can explan to me, how can i use the dev branch,
thx a lot

MartiPri, 2022-11-23:
> Hi, i'm the next guy, how has the same issue with Ketotek _TZE200_bvu2wnxz :-( My HA Zigbee2MQTT shows "unsupported". Who can explan to me, how can i use the dev branch, thx a lot

Hey 👋🏻 . I'm already using the dev branch. It also doesn't work with that in conjunction with the Ketotek. 😉

MartiPri, 2022-11-23:
Next radiator device not supported 😑 . _TZE200_3yp57tby  by MOLO. One after another I buy there are all not supported in dev branch.

mikeg-de, 2022-11-23:
FYI – Some level of conversation about the same converted is happening in https://github.com/Koenkk/zigbee2mqtt/issues/15009#issuecomment-1324892985. I noticed, though, that the converter seems to lack some exposed settings like schedule for the day of the week, hour etc..

Might be worth continuing in the other thread.

HumanSkunk, 2022-11-24:
I'll add another one to the Ketotek variant of the TS0601 as well. Got it on Amazon during the sale and thought it was just a Tuya effort and should work as my other TRVs seem the same. Has anyone managed to get it working? Read through the add a new device using converters but havent had any time to look through any of it yet.

aleslukek, 2022-12-02:
I have bunch of TZE200_3yp57tby radiator devices. Is there a way to manually include them somewhere to get them supported?

AlpineP, 2022-12-26:
I have the same device as MartiPri & aleslukek.
The TZE200_3yp57tby. I can connect to it, but the temperatures and batteri level are not showen. Any fix for this?

charly50, 2022-12-26:
i switch to  Unitec / TS0601 / _TZE200_ps5v5jor and it works